ChefKissInc/NootedRed #267: AMD laptop with 4K screen replacement, black screen after sleep and wake-up.

Paulchan82, 2024-05-25:
### macOS Version

Sonoma

### What is your CPU's model?

AMD 5900HX

### Please describe the behaviour in detail.

AMD 5900hx, using nootedred driver for integrated graphics, the original screen is 2k 165hz, and it can light up after sleep and wake up.

Due to a damaged screen and color distortion, a 4k 120hz screen from AUO was replaced. The retina screen effect has indeed improved significantly, but after sleep and wake up, the screen only lights up for one second and then goes black. The keyboard and mouse respond, and remote access to the desktop is possible, but a forced shutdown and reboot are required.

Sleep and wake up under Windows are completely normal, indicating that it is not a problem with the screen.

### What should've happened instead?

please fix it,thx

Paulchan82, 2024-05-25:
The -NRedDPDelay parameter has been used, but the screen remains black. The VRAM has been adjusted to 4GB in the BIOS.

VisualEhrmanntraut, 2024-06-03:
I had a bug with my workflow and the bot did not reply to the issue with the first interaction greeting:

> Greetings. It seems like this is the first issue you open on this repository.
> 
> We are letting you know that these are for bug reports or feature requests. Most of the reports we receive in this GitHub Organisation are user errors.
> 
> For the sake of saving time, here are the most common cases:
> - Outdated version of this kext.
> - Conflicts with other kexts (e.g. `WhateverGreen`).
> - Issues caused by other kexts like memory corruption (e.g. `AirportItlwm`).
> - Intended behaviour by Apple (e.g. V-Sync or cursor rendering).
> - Using cheap hacks like kext/library downgrades (e.g. `OCLP`, `BFixup`).
> - Duplicate kexts.
> - Malformed config.plist caused by configurator software (`OCAT`, `OCC`, etc).
> - Lack of TSC synchronisation (use `AmdTscSync`).
> - Incorrectly configured TSC synchronisation (set `IOCPUNumber` in the `Info.plist` to number of threads minus one for macOS 11 and older, 0 for anything newer).
> - Outdated macOS minor version (e.g. macOS 11.0.0 instead of latest).
> - Using beta versions of macOS.
> - Using the kext during installation or update. There is a Lilu bug which makes macOS stall during the last install phase.
> - Incorrect SMBIOS. Please use `iMac20,1`, `MacBookPro16,3`, `MacPro7,1` or `iMacPro1,1`.
> - (e)DP link training failure. `-NRedDPDelay` will fix this as mentioned in the README. This is last resort, black screen can be caused by other factors mentioned.
> - Insufficient VRAM size.
> 
> We will never support the use of configurator software, solutions like OCLP/BFixup, or using macOS beta versions.
> 
> This issue will be looked over by the respective maintainer when they can. In the meantime, look if you can resolve this yourself via checking the above. 
> 
> Be patient, we are hobbyists.

VisualEhrmanntraut, 2024-07-23:
May be caused by #284

AlfCraft07, 2024-07-23:
Mine works but the keyboard dies after wake

VisualEhrmanntraut, 2024-07-23:
@AlfCraft07 That happens to my testing laptop too sometimes, but sometimes the screen does not wake, and I have a suspicion that one it might be because of the faulty firmware injection

VisualEhrmanntraut, 2024-11-05:
As mentioned above, I theorise that it's because of #284, so merging this issue with it.

closed 2024-11-05 by VisualEhrmanntraut