ChefKissInc/NootedRed #268: Many apps crash on ASUS Vivobook with 7530U and Ventura

Bradamante3D, 2024-05-26:
### macOS Version

Ventura

### What is your CPU's model?

AMD Lyzeen 7530U

### Please describe the behaviour in detail.

Hi there,

first of all thank you a lot for NootedRed, it's probably one of the most exciting Hackintosh projects in the space right now, since it allows us to use more modern hardware, newer even than Apple's cutoff date. I bought a ASUS Vivobook 16 with a 7530U (Vega 7). Getting Ventura to run was easier than expected and the performance feels good.

Unfortunately, the notebook is almost unusable since many apps just crash on startup. Since I don't have much experience with AMD Hackintoshes (but a lot with Intel based ones), I can't even be certain if it's NootedRed or my bad EFI config. Since a lot of things work, and since I can see graphics driver crashes in the logs, I assume however that it's NR.

I am trying to attach crash logs and my EFI. You will notice that I experimented a lot with GPIO, since the trackpad is currently not recognized. Ignore that. I also haven't tested much:

WHAT WORKS
QuickTime - Can play a 4K 60 FPS HDR video just fine.
Safari - Can watch YouTube videos just fine.
League of Legends (client) - works.
Unity "Book of the Dead" demo - That's a Metal export of the "Book of the Dead" demo that Unity put out in 2018, I am using it often to test new builds.
Apps like Hackintool, MaciASL, Photos, BBEdit, OpenCore Configurator.

WHAT DOESN'T WORK
Photo Booth - Content of webcam is shown fine, app can also be quit. But crashes as soon as you try to resize the window.
VLC - Audio for video playback works, but video window turns red and crashes.
Firefox, Thunderbird - Crash on startup.
League of Legends (game) - Crash on startup.
Heaven (benchmark) - Client works, but benchmark itself crashes on startup.
Interestingly, the crashes feel all the same. The entire machine freezes for maybe 20-30 seconds. Mouse pointer still moves, but Finder or Dock are frozen. Then, the machine comes back with a crash message.

I would love to provide more crash logs, but for a lot of apps I can't figure out where to find them.

Would be cool if you find the attached logs useful. I will also send you some more money via Paypal. Thank you for all your work.
[20250525 Vivobook 7530U.zip](https://github.com/ChefKissInc/NootedRed/files/15447639/20250525.Vivobook.7530U.zip)

### What should've happened instead?

Not crash.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[20250525 Vivobook 7530U.zip](https://github.com/ChefKissInc/NootedRed/files/15447643/20250525.Vivobook.7530U.zip)

wajdi-coder, 2024-05-26:
How many gb of vram do you set ? I faced the same problem previously It turned out later that I was using 512MB for vram and i couldn't increase it from bios then i used a tool from github that allows to increase it to 2GB . This is the link for that tool if you are interested .
https://github.com/DavidS95/Smokeless_UMAF
i am sorry about my poor english .

VisualEhrmanntraut, 2024-05-26:
Hi.

First and foremost, since you mentioned using OCC, we disapprove and do not support the usage of configurator tools because of the following:

1. They tend to break the config.plist or generally don't work all that great
2. They don’t respect kext and tool order
3. They encourage being lazy

A lot of Hackintosh developers, including a bunch of Acidanthera's members (I hope that's not hearsay), don't like these kinds of tools either.

Second of all, what you are reporting is the already known OpenGL issues (Chromium, LoL, etc all use OpenGL).

Furthermore, issues with applications such as Firefox have already been mostly resolved.

As you still have issues with them, it's likely one of the following:

- VRAM size is insufficient
- A configuration issue is causing problems like memory corruption
- Some other kexts is causing some kind of conflict

I would like to remind you that this is just my hobby, and as of recent times I do not have the time or strength like the beginning of the project due to personal issues (work, preparing to move to another country, among other issues).

I see a lot of issues with the configuration you posted, it would be too much work to list them all. I suspect you either got and/or edited an EFI from the internet, or someone else from the people that make EFIs completely wrong has heavily influenced the creation of this EFI. I'd recommend you remake it following my guide which is at https://chefkissinc.github.io/guide and then the Ryzen config.plist guide in Dortania (dortania.github.io is the website, there's a lot of fake ones so be careful), since my guide has not gotten the config.plist sections completed yet.

I hope that you do not post any negative response to this and make my life harder than it already is. I'm already barely able to keep at it with Hackintosh work at this point.

Bradamante3D, 2024-05-26:
Hi there, thank you for your feedback. I find your response unnecessarily negative, I have already praised and thanked you for your work, and I meant it, and I tried to express my gratitude financially.

I agree that lack of assigned VRAM is a likely cause, but I have not seen an option in the BIOS to increase it. I have seen Smokeless_UMAF, but I refrained from using it, due to fear of bricking my hardware. I will take a second look at it. 

I am aware that OpenCore Configurator is frowned upon by some, but I have found it useful and never had problems with it. I will use a different tool to double check my EFI, to see if OCC has screwed something up.

I have not used a different guide found elsewhere, I did follow the official Dortania guide as well as your own guide. I can go through the official Dortania guide again, to see if I did something wrong in my EFI, but I doubt it. I have been in the hobby since 2012, and do extensive online research myself, instead of throwing a crash report at somebody with a "please fix" attitude. As I said in my original post, the problems you see in my EFI are probably remnants of my experimentation to get the trackpad to work.

I completely respect your lack of time for this hobby project, I can relate. I did not post my crash logs expecting an immediate investigation and quick fix, I posted them to help as much as I can, since I know from my own experience in IT that vague complaints are not helpful.

VisualEhrmanntraut, 2024-05-27:
Bricking your machine with this tool is not possible, only soft bricking which can be resolved by clearing CMOS. As long as you don't mess with any weird setting, you save then hit "reset" in the main menu (not "continue") then nothing will happen.

Bradamante3D, 2024-05-27:
Good news. I just used Smokeless_UMAF to increase the VRAM to 4 GB and now Thunderbird, Firefox, VLC work as expected. I have not tested more yet. Quick instructions:
1) Go to https://github.com/DavidS95/Smokeless_UMAF Scroll down a bit, there is a Download section that lets you download: UniversalAMDFormBrowser
2) In MacOS's disc utility (or equivalent tool), format a USB stick with FAT (not: exFAT) and Master Boot Record
3) Move the extracted ZIP content to the USB stick. Your file structure should look like in one of the screenshots on Smokeless_UMAF's Github page
4) Boot from the USB stick (can for example be done by using the Boot Override option in BIOS)
5) Go to: Device Manager -> AMD CBS -> NBIO Common Options -> GFX Configuration -> iGPU Configuration. Set from Auto to UMA_SPECIFIED. A new option UMA Frame buffer size appears. Set the size you want.

VisualEhrmanntraut, 2024-05-27:
Nice. Another tip that's not in the site yet is that you should try out my new TSC kext ForgedInvariant instead, see if that helps your system in general. Have fun.

closed 2024-05-27 by VisualEhrmanntraut