ChefKissInc/NootedRed #269: No Artifacts Generated

Jingexz01, 2024-05-29:
### macOS Version

Big Sur

### What is your CPU's model?

AMD A8-7680

### Please describe the feature in detail.

I can't download nootedred as there are no artifacts generated in actions tab, already searched but I really can't download. Artifact is there but it is blanked out "-"

### What alternatives have you tried, if any?

None

techmare, 2024-05-29:
@Jingexz01 Hey, I just wanna point out that nootedred don't support your cpu ! Nootedred supports the Ryzen series up to 5th generation

VisualEhrmanntraut, 2024-05-30:
that and releases tab

Jingexz01, 2024-05-30:
hey! does that mean I cant use nootedred on my integrated gpu even with gpu spoofing? its just very annoying to use macOS and the visuals are laggy because it is only using 4mb vram, are there any other alternatives or patches to enable my pc to use more than 4mb ram to fix the laggy visuals? Thanks!

techmare, 2024-05-30:
@Jingexz01 No 👎 it's won't work :/

VisualEhrmanntraut, 2024-05-30:
there is no working kext for these old APUs. ChefKissInc/LegacyRed only can provide framebuffer as of now.

closed 2024-05-30 by VisualEhrmanntraut

Jingexz01, 2024-05-30:
for LegacyRed, do I need to put any boot-args for it to work or would it work right away after being placed in kexts folder? And also would it work alongside Whatevergreen?