ChefKissInc/NootedRed #270: Black screen on internal display

win8bugpas, 2024-06-02:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 7 3700U with Radeon Vega 10

### Please describe the behaviour in detail.

When NootedRed is enabled, only the external monitor works, and the graphics acceleration works fine. But the internal display stays completely black.
The screen works when NootedRed is disabled but without GPU acceleration.
I use the latest commit of NootedRed.

### What should've happened instead?

Both displays working

tlfares, 2024-06-03:
> Put the laptop in sleep mode, then bring it out and see 2 active screens. Or try to activate 2 monitors via a third-party application. This is also the case with real macintosh. ¯_(ツ)_/¯

I tried but after I reopened the laptop, the laptop screen is still off :/
What third party application can I try to activate it?

tlfares, 2024-06-03:
In the system report, the internal screen is not even detected, only the external monitor
[image]

VisualEhrmanntraut, 2024-06-03:
I had a bug with my workflow and the bot did not reply to the issue with the first interaction greeting:

> Greetings. It seems like this is the first issue you open on this repository.
> 
> We are letting you know that these are for bug reports or feature requests. Most of the reports we receive in this GitHub Organisation are user errors.
> 
> For the sake of saving time, here are the most common cases:
> - Outdated version of this kext.
> - Conflicts with other kexts (e.g. `WhateverGreen`).
> - Issues caused by other kexts like memory corruption (e.g. `AirportItlwm`).
> - Intended behaviour by Apple (e.g. V-Sync or cursor rendering).
> - Using cheap hacks like kext/library downgrades (e.g. `OCLP`, `BFixup`).
> - Duplicate kexts.
> - Malformed config.plist caused by configurator software (`OCAT`, `OCC`, etc).
> - Lack of TSC synchronisation (use `AmdTscSync`).
> - Incorrectly configured TSC synchronisation (set `IOCPUNumber` in the `Info.plist` to number of threads minus one for macOS 11 and older, 0 for anything newer).
> - Outdated macOS minor version (e.g. macOS 11.0.0 instead of latest).
> - Using beta versions of macOS.
> - Using the kext during installation or update. There is a Lilu bug which makes macOS stall during the last install phase.
> - Incorrect SMBIOS. Please use `iMac20,1`, `MacBookPro16,3`, `MacPro7,1` or `iMacPro1,1`.
> - (e)DP link training failure. `-NRedDPDelay` will fix this as mentioned in the README. This is last resort, black screen can be caused by other factors mentioned.
> - Insufficient VRAM size.
> 
> We will never support the use of configurator software, solutions like OCLP/BFixup, or using macOS beta versions.
> 
> This issue will be looked over by the respective maintainer when they can. In the meantime, look if you can resolve this yourself via checking the above. 
> 
> Be patient, we are hobbyists.

tlfares, 2024-06-03:
> I had a bug with my workflow and the bot did not reply to the issue with the first interaction greeting:

Thx for your reply, I saw this in other issues threads and I already tried all of that 🥲

VisualEhrmanntraut, 2024-06-03:
The greeting in the other issues is out of date in the TSC sync part.

VisualEhrmanntraut, 2024-06-03:
I sent the outdated text by accident. Check it again.

tlfares, 2024-06-03:
Oh ok, I replaced it with ForgedInvariant but I still have the same issue

arquicion333, 2024-06-07:
Hello! I have a similar issue too.
I've installed Ventura on my Dell Inspiron 3525 that works with a Ryzen 5 5625U and Radeon Vega 7. I having no luck with the internal display. It does not work, it has not signal at all most the time. I managed to end ud with the installation using and external monitor (hdmi). Ive already tried so many things. Debugs kexts, DSDT patches, etc... Some times the external display shows up as the second monitor and the internal display shows some black light. I could share my EFI or the Ioreg.
The internal display is not even detected in the ioregistry. 
I may think it is relate to the iGPU EDID, but I just don't what to do with it. 
The Bios Setup of this Laptop is really basic. I'd read that it is necessary to increase the VRAM in order to have a proper performance but there is no way to change that using the bios setup. System Report shows the iGPU as Radeon Graphics 512mb.
I'd checked all the stuff you marked on common cases with no luck. :(

tlfares, 2024-06-08:
>I'd read that it is necessary to increase the VRAM in order to have a proper performance but there is no way to change that using the bios setup.

I've tried to increase the VRAM set by the BIOS to 1 or 2Gb (using this tool https://github.com/DavidS95/Smokeless_UMAF) but that didn't changed anything for me 🥲

arquicion333, 2024-06-08:
> I've tried to increase the VRAM set by the BIOS to 1 or 2Gb (using this tool https://github.com/DavidS95/Smokeless_UMAF) but that didn't changed anything for me 🥲

I didn't try increasing de VRAM, because that is mostly related to the IGPU performance. This issue is related to the EDP (Embedded Display Port), or the EDID. Someone suggested me to use the Smoless_UMAF tool to change some Edp settings. I'm running the Smokeless_UMAF tool right now. At the PBS, I only have 2 options related: DP0, where I can choose EDP, DP or HDMI, and DP1, where I can choose DP or HDMI. There are no others settings related to the EDP that I can see. By default, the EDP is assigned to DP0. I guess that changing these variables won't fix the problem. And I'm a little afraid of missing the regular bios, because of the warnings related to dell's laptop users. I'd read about the fix for Dell's laptop but I found no way to delete Misc options by accessing to the tool's Boot Manager... Anyways, I think I could re-flash the chip bios with a programmer if something wrong happens. 

Any ideas?

techmare, 2024-06-08:
@tlfares Boot windows and go to ur task manager and then performance and then ram and check how much of your ram is reserved. If it is 2 GB then u have to 2 GB of VRAM

arquicion333, 2024-06-09:
> @tlfares Boot windows and go to ur task manager and then performance and then ram and check how much of your ram is reserved. If it is 2 GB then u have to 2 GB of VRAM

Thanks, already did that! I can handle easily 4GB of VRAM cause the laptop have 16GB of RAM... Unfortunately it didn't work. I believe I have to create some special DSDT to have the EDP working, or create devices properties to address the EDID properly. 
I'm just burned out with this issue, but I'm not giving up. I will report myself later! Thanks, mate!

Tang061022Tang, 2024-06-13:
Have you ever tried the dpdelay nvram?

Tang061022Tang, 2024-06-13:
Have you ever tried to add the dpdelay command into boot-args?

arquicion333, 2024-06-13:
> Have you ever tried to add the dpdelay command into boot-args?

Hola amigo, 
Well, I did not, how would it be? I've found the internal display now, and the other connector. They go through the AMDRadeon kexts a frame buffer. But I can't figure it out how to creat a DSDT, or to make a custom EDID.
Can you tell me about dp command into boot args? I will research about it. Thanks!

arquicion333, 2024-06-13:
> Have you ever tried to add the dpdelay command into boot-args?

God yeah! -NRedDPDelay works flawlessly! Don't how I made it to ignore this point of the troubleshooting. It must be the devil...
Muchas gracias, amigo!

tlfares, 2024-06-13:
> @tlfares Boot windows and go to ur task manager and then performance and then ram and check how much of your ram is reserved. If it is 2 GB then u have to 2 GB of VRAM

Yes for the dedicated GPU memory, it shows 2Gb

Trijal08, 2024-08-01:
@VisualEhrmanntraut Thank you so much for putting this request into consideration! I appreciate that!

arquicion333, 2024-08-01:
Hello, brother my brother,

I appreciate all of your workship.
I also have another new laptop now, an LG Ultra 16UD70R-G.AX59B. 
Ryzen zen4 7000 series on it.  With Vega 6 on it.

I make it work almost flawlessly, but I could not have the internal display to work properly. Black screen. The only fix so far: I have to close the display for 7 seconds, then I open it up and voila!

Many thanks for so much good work!

Cheers!

Trijal08, 2024-08-03:
@VisualEhrmanntraut I am certain that some of the commit(s) **after** [this one](https://github.com/ChefKissInc/NootedRed/commit/ede4b28d82c67bbe40ae63b4df3a053fe8b00fa1) caused the Internal display issue on some Vega 10 APUs. I remember that this commit worked for me on Ventura as I have stated in my older issue [here](https://github.com/ChefKissInc/NootedRed/issues/181); but then when I tried to use an updated version of it, it suddenly started showing a black screen on boot, and never stopped ever after. And also, I tired using that old build but failed to do so on any major version newer than macOS Ventura. I hope this helps you debug and resolve this matter quicker! 😄

Trijal08, 2024-08-03:
@VisualEhrmanntraut I am certain that some commit(s) **after** [this one](https://github.com/ChefKissInc/NootedRed/commit/ede4b28d82c67bbe40ae63b4df3a053fe8b00fa1) caused and still causes the internal display issue on some Vega 10 APUs. As I said in my previous problem [here](https://github.com/ChefKissInc/NootedRed/issues/181), I seem to recall that this commit worked for me on Ventura. However, when I tried to use an updated version of it; the internal display suddenly started showing a black screen on boot, and never stopped ever after. Also, I tried using the older build but failed to do so as it refuses to even boot any major version newer than macOS Ventura. Furthermore, I have tried to use an external display using the HDMI port with the latest 1.0.0 build, but again, failed to do so. I hope this helps you pinpoint and resolve this issue quicker! 😄

Trijal08, 2024-08-03:
@VisualEhrmanntraut I am certain that some commit(s) **after** [this one](https://github.com/ChefKissInc/NootedRed/commit/ede4b28d82c67bbe40ae63b4df3a053fe8b00fa1) caused and still causes the internal display issue on some AMD Radeon Vega 10 iGPU laptops. As I said in my previous problem [here](https://github.com/ChefKissInc/NootedRed/issues/181), I seem to recall that this commit worked for me on Ventura. However, when I tried to use an updated version of it; the internal display suddenly started showing a black screen on boot, and never stopped ever after. Also, I tried using the older build but failed to do so as it refuses to even boot any major version newer than macOS Ventura. Furthermore, I have tried to use an external display using the HDMI port with the latest 1.0.0 build, but again, failed to do so. I hope this helps you pinpoint and resolve this issue quicker! 😄

VisualEhrmanntraut, 2024-08-03:
@Trijal08 you sending and constantly deleting that comment to make edits to it is making my computer and my phone's notifications go nuts, both on Telegram, my email, and the GitHub Mobile app. Stop!!! There's an edit button if you click the dots on the upper right of the comment.
[image]

Trijal08, 2024-08-03:
> @Trijal08 you sending and constantly deleting that comment to make edits to it is making my computer and my phone's notifications go nuts, both on Telegram, my email, and the GitHub Mobile app. Stop!!! There's an edit button if you click the dots on the upper right of the comment.
> [image]
> 

Sorry man, I didn't know that'll happen. Next time, I'll edit instead.

Trijal08, 2024-08-04:
> In the system report, the internal screen is not even detected, only the external monitor [...]

Not even external display works for me.
☠️

VisualEhrmanntraut, 2024-08-04:
@Trijal08 If the link training fails the display won't show up in settings. You mentioned there was a point where it was working. You need to find the exact commit that broke your display signal. I can't find this out myself, as mentioned in my screenshot, because I obviously don't have one of these machines.

Trijal08, 2024-08-04:
> @Trijal08 If the link training fails the display won't show up in settings. You mentioned there was a point where it was working. You need to find the exact commit that broke your display signal. I [...]

This: https://github.com/ChefKissInc/NootedRed/commit/86bd323c104b27988db41f2a31211d8e6465eb93
Can you maybe somehow try reverting that commit and give me the newly generated build artifacts? I would really appreciate it! However, if that doesn't work, I'll try doing something on my end and make PR in your repo if possible. I really do appreciate your work and would apprecate it even more if you could accomplish this feet!

VisualEhrmanntraut, 2024-08-04:
There's no way that commit broke your display, that just moved the HDMI audio from being injected always to only on main OS.

VisualEhrmanntraut, 2024-08-04:
I have some other suspicions though. 

Try these builds and tell me which one works.
[NootedRed-1.0.0-RELEASE.zip](https://github.com/user-attachments/files/16488692/NootedRed-1.0.0-RELEASE.zip)
[NootedRed-1.0.0-RELEASE.zip](https://github.com/user-attachments/files/16488694/NootedRed-1.0.0-RELEASE.zip)

Trijal08, 2024-08-04:
> I have some other suspicions though.
>
> Try these builds and tell me which one works. [NootedRed-1.0.0-RELEASE.zip](https://github.com/user-attachments/files/16488692/NootedRed-1.0.0-RELEASE.zip) [...]

OMG! I FIXED IT! So basicallly I forked youe repository and added those deletions back; and voila, my internal display is working like a charm and the graphics are accelerated too! Your builds didn't do shiiteza! Here is my build if you want to test it or something: https://github.com/Trijal08/NootedRed/actions/runs/10238799363
The commit I made is here: https://github.com/Trijal08/NootedRed/commit/56b47d7c73fe23b59c05bbd77550bfbec165f3f7
If you'd like, I can make a PR.

Trijal08, 2024-08-04:
> There's no way that commit broke your display, that just moved the HDMI audio from being injected always to only on main OS.

Nevermind, it broke again. How sad is that!?

VisualEhrmanntraut, 2024-08-04:
Yeah uh how about I just block you from the organisation instead because you are overconfident and annoying?
(reader's context: he was also just as overconfident and annoying when he was in the Telegram group)

Trijal08, 2024-08-04:
> Yeah uh how about I just block you from the organisation instead because you are overconfident and annoying? (reader's context: he was also just as overconfident and annoying when he was in the Telegram group)

I apologize. But how is that even possible? How does it work then it doesn't?

VisualEhrmanntraut, 2024-08-04:
If your parents told you "if you do something wrong, just say sorry, it makes everything better", that's not how it works.

Trijal08, 2024-08-04:
> If your parents told you "if you do something wrong, just say sorry, it makes everything better", that's not how it works.

Okay, I get it. I'll try not to be "overconfident and/or annoying" to you or somebody in this organization.

VisualEhrmanntraut, 2024-08-04:
No. I suggest you chill with your internet usage and spend more time outside to improve your mental health. I don't want to handle a child having a mrbeast-type youtuber reaction to everything related to the projects. Thanks and wish you a better state of mind.

VisualEhrmanntraut, 2024-10-08:
Maybe now that I put back the old brightness logic it magically fixes this issue. Can you all retry?

Kfkcome, 2024-10-09:
Sir,is your external monitor connected by VGA or HDMI? When I connect external monitor by VGA ,it comes to a blue screen and cannot display anything else.

VisualEhrmanntraut, 2024-10-09:
What, me? I use an M3 MacBook Pro, lol...

arquicion333, 2024-10-10:
> Sir,is your external monitor connected by VGA or HDMI? When I connect external monitor by VGA ,it comes to a blue screen and cannot display anything else.
Hello, VGA is an output unfamiliar for Macs, it is rarely possible to have it working.

Kfkcome, 2024-10-11:
> > Sir,is your external monitor connected by VGA or HDMI? When I connect external monitor by VGA ,it comes to a blue screen and cannot display anything else.
> > Hello, VGA is an output unfamiliar for Macs, it is rarely possible to have it working.

Thanks, it works fine now. I buy a new VGA to HDMI adapter.

Otus9051, 2025-02-07:
@tlfares Did your device happen to have a touchscreen?
I am on a ThinkPad T495 with a R7 3700U and am facing the same issue of the display just being black. HDMI out doesn't work, neither the port nor from USB C

I theorize this on the basis that the T495 touchscreen seems to have a different cable (https://www.reddit.com/r/thinkpad/comments/hxl8e5/experiment_on_upgrading_t495s_screen_to_wqhd/) than the normal ones, and maybe the pinout is weird and prohibits NRed from identifying the hardware/working?

Otus9051, 2025-02-14:
Pinging this issue again.

VisualEhrmanntraut, 2025-02-17:
> Pinging this issue again.

That will just result in me locking it and unlocking it when there is an update :^)

Otus9051, 2025-02-19:
@VisualEhrmanntraut In my opinion, that would be a better decision. Or you can entirely drop support for these systems and mention it in the README. Probably will attract more people to try fix it.

VisualEhrmanntraut, 2025-02-19:
It's not certain which laptops get a black screen issue or not, if that was known, I would know the source of the issue.

Otus9051, 2025-02-19:
That's what I am trying to figure out. A different pinout might be the issue as far as I know. I can provide logs if you want.

Otus9051, 2025-02-19:
T495 seems to have 2 lane graphic + in-cell touch for the touchscreen and the normal one has 4 lanes, maybe that could cause bandwidth limitations?
Source: https://www.reddit.com/r/thinkpad/comments/hxl8e5/experiment_on_upgrading_t495s_screen_to_wqhd/

VisualEhrmanntraut, 2025-03-22:
Please retry with latest commit.