ChefKissInc/NootedRed #271: macOS Sequoia compatibility

AlfCraft07, 2024-06-10:
### macOS Version

Sonoma

### What is your CPU's model?

AMD 3020e (the model proposed by the example lol)

### Please describe the feature in detail.

Now that macOS Sequoia was announced, when will NootedRed get compatibility for it? I'm really excited to try it, and I can do that only on this PC.

### What alternatives have you tried, if any?

I didn't try anything, as I know macOS Sequoia is just 1 hour old and kext developers need time to make everything compatible with it.

VisualEhrmanntraut, 2024-06-10:
you will have to be in the telegram group to be a lab rat for this

VisualEhrmanntraut, 2024-06-10:
actually it's impossible right now because Lilu has not even been updated to support it yet, and there is no KDK out afaik so if the current patches aren't compatible it also means they can't be updated

AlfCraft07, 2024-06-11:
If you look into the hackintosh subreddit you can find people that managed to set up Sequoia hackintoshes on CPUs starting with 4th gen intel