ChefKissInc/NootedRed #282: Cannot play video which is downloaded in Sequoia

Kfkcome, 2024-07-19:
### macOS Version

Sonoma

### What is your CPU's model?

AMD R7-4800H

### Please describe the behaviour in detail.

Cannot play video which is downloaded.The video stuck all the time like a pause, but the play button has been pressed.I have already disable the hardware acceleration.

[image]
(IINA video player)

### What should've happened instead?

I used many video player but all of them are in the same. When I used Sonoma , it was fine . It can play the video which is downloaded soomthly.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

If it is needed, I can upload.

VisualEhrmanntraut, 2024-07-19:
AppleALC broken in Sequoia = no sound = video playback fails

Kfkcome, 2024-07-19:
> AppleALC broken in Sequoia = no sound = video playback fails

Yeah, it is true that appleAlC broken in my laptop . You are right , When I use the bluetooth earphone , it works fine.Thanks bro.

closed 2024-07-19 by VisualEhrmanntraut