ChefKissInc/NootedRed #283: Kernel pani with NootedRed on ASUS vivobook with 7530U

musicthebee, 2024-07-22:
### macOS Version

Ventura

### What is your CPU's model?

Ryzen 7530U

### Please describe the behaviour in detail.

I was able to successfully install Ventura on ASUS vivobook with Ryzen 7530U. However upon replacing WhateverGreen with NootedRed I get Kernel Panic. Here is the panic log from the OS. I have used vanilla opencore with all the patches manually created going through the guide located at https://chefkissinc.github.io/guide. Any help is appreciated. Thank you for your great work!!!

{"bug_type":"210","timestamp":"2024-07-21 17:12:09.00 -0700","os_version":"macOS 13.2.1 (22D68)","roots_installed":0,"incident_id":"EE9EA5E3-9339-46F7-9D83-074958C68562"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBwYW5pYyBzdGFja3Nob3QgZm91bmQgaW4gY29yZWZpbGU=","macOSPanicString":"panic(cpu 0 caller 0xffffff8016a1020a): NootedRed   X6000FB: @ Failed to apply logic revert patches\n\nPanicked task 0xffffffa10c3c5208: 157 threads: pid 0: kernel_task\nBacktrace (CPU 0), panicked thread: 0xffffffaf70675598, Frame : Return Address\n0xffffff9a84143670 : 0xffffff8010feb38d mach_kernel : _handle_debugger_trap + 0x4ad\n0xffffff9a841436c0 : 0xffffff8011158ed6 mach_kernel : _kdp_i386_trap + 0x116\n0xffffff9a84143700 : 0xffffff8011148120 mach_kernel : _kernel_trap + 0x3e0\n0xffffff9a84143750 : 0xffffff8010f85951 mach_kernel : _return_from_trap + 0xc1\n0xffffff9a84143770 : 0xffffff8010feb66d mach_kernel : _DebuggerTrapWithState + 0x5d\n0xffffff9a84143860 : 0xffffff8010fead19 mach_kernel : _panic_trap_to_debugger + 0x1a9\n0xffffff9a841438c0 : 0xffffff80117e072b mach_kernel : _panic + 0x84\n0xffffff9a841439b0 : 0xffffff8016a1020a com.ChefKissInc.NootedRed + 16906\n0xffffff9a84143bc0 : 0xffffff8016a1422a com.ChefKissInc.NootedRed + 33322\n0xffffff9a84143ca0 : 0xffffff80169abfdd as.vit9696.Lilu : __ZN7LiluAPI20onEntitlementRequestEPFvPvP4taskPKcRP8OSObjectES0_ + 0x27d\n0xffffff9a84143ce0 : 0xffffff80169a3e75 as.vit9696.Lilu : __ZN13KernelPatcher18setupKextListeningEv + 0x1835\n0xffffff9a84143d30 : 0xffffff8011670379 mach_kernel : __ZN6OSKext4loadEhhP7OSArray + 0x729\n0xffffff9a84143d90 : 0xffffff801168683e mach_kernel : __ZN6OSKext22loadKextWithIdentifierEP8OSStringPP8OSObjectbbhhP7OSArray + 0x1ae\n0xffffff9a84143de0 : 0xffffff801170ad2e mach_kernel : __ZNK11IOCatalogue14isModuleLoadedEP12OSDictionaryPP8OSObject + 0xee\n0xffffff9a84143e30 : 0xffffff80116e7abf mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0x59f\n0xffffff9a84143ef0 : 0xffffff80116e7390 mach_kernel : __ZN9IOService14doServiceMatchEj + 0x3b0\n0xffffff9a84143f50 : 0xffffff80116ea497 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x157\n0xffffff9a84143fa0 : 0xffffff8010f8519e mach_kernel : _call_continuation + 0x2e\n      Kernel Extensions in backtrace:\n         as.vit9696.Lilu(1.6.7)[087C34D2-49F8-3FDA-8757-27E9425C9EE4]@0xffffff80169a1000->0xffffff80169ccfff\n         com.ChefKissInc.NootedRed(1.0)[F0064AC6-5406-384B-AF9A-87C8F2330736]@0xffffff8016a0c000->0xffffff8016c58fff\n            dependency: as.vit9696.Lilu(1.6.7)[087C34D2-49F8-3FDA-8757-27E9425C9EE4]@0xffffff80169a1000->0xffffff80169ccfff\n            dependency: com.apple.iokit.IOPCIFamily(2.9)[802B9CA4-B851-396E-8DA4-B30E751EC48F]@0xffffff8013a59000->0xffffff8013a88fff\n\nProcess name corresponding to current thread (0xffffffaf70675598): kernel_task\nBoot args: -v keepsyms=1 debug=0x100 npci=0x3000 alcid=1 \n\nMac OS version:\n22D68\n\nKernel version:\nDarwin Kernel Version 22.3.0: Mon Jan 30 20:42:11 PST 2023; root:xnu-8792.81.3~2\/RELEASE_X86_64\nKernel UUID: 10E5D254-4A37-3A2A-B560-E6956A093ADE\nroots installed: 0\nKernelCache slide: 0x0000000010c00000\nKernelCache base:  0xffffff8010e00000\nKernel slide:      0x0000000010cdc000\nKernel text base:  0xffffff8010edc000\n__HIB  text base: 0xffffff8010d00000\nSystem model name: iMac20,1 (Mac-CFF7D910A743CAAF)\nSystem shutdown begun: NO\nPanic diags file available: YES (0x0)\nHibernation exit count: 0\n\nSystem uptime in nanoseconds: 10994369513\nLast Sleep:           absolute           base_tsc          base_nano\n  Uptime  : 0x000000029011a6ac\n  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000\n  Wake    : 0x0000000000000000 0x000000090d19385c 0x0000000000000000\nCompressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space\nZone info:\n  Zone map: 0xffffff9aa425e000 - 0xffffffbaa425e000\n  . PGZ   : 0xffffff9aa425e000 - 0xffffff9aa625f000\n  . VM    : 0xffffff9aa625f000 - 0xffffff9f72a5e000\n  . RO    : 0xffffff9f72a5e000 - 0xffffffa10c25e000\n  . GEN0  : 0xffffffa10c25e000 - 0xffffffa5d8a5e000\n  . GEN1  : 0xffffffa5d8a5e000 - 0xffffffaaa525e000\n  . GEN2  : 0xffffffaaa525e000 - 0xffffffaf71a5e000\n  . GEN3  : 0xffffffaf71a5e000 - 0xffffffb43e25e000\n  . DATA  : 0xffffffb43e25e000 - 0xffffffbaa425e000\n  Metadata: 0xfffffff5f0447000 - 0xfffffff610447000\n  Bitmaps : 0xfffffff610447000 - 0xfffffff616447000\n\n\n\n"}

### What should've happened instead?

I should have been able to boot to OS successfully.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

Attached panic file

Zormeister, 2024-07-22:
Your macOS copy needs updating.

musicthebee, 2024-07-22:
Thanks a lot. That worked

closed 2024-07-22 by musicthebee