ChefKissInc/NootedRed #285: rDNA 2 Laptop

iltechalexone, 2024-07-26:
### macOS Version

None

### What is your CPU's model?

AMD Ryzen 7 6800h

### Please describe the feature in detail.

Hello, when will be available SUPPORT for rDNA 2 Laptops? I have a Ryzen 7 6800h, with a 680m rembrandt...

PLEASE ADD THIS :)

### What alternatives have you tried, if any?

.

closed 2024-07-26 by VisualEhrmanntraut

MiranSS02, 2024-09-27:
is working on the 680M project abandoned now?

VisualEhrmanntraut, 2024-09-27:
Hi. It’s not abandoned, just lower priority. I am the developer of all these kexts, I can’t magically just get stuff working. Just wait.

MiranSS02, 2024-09-27:
Sorry to bother you, i was just curios since it's a great IGPU with that performance. hope we see it soon.
Thanks for your efforts.

VisualEhrmanntraut, 2024-09-27:
You're welcome.