ChefKissInc/NootedRed #290: Brightness control work in HDR, but not in SDR (Renoir)

RasyiidWho, 2024-08-13:
### macOS Version

Ventura

### What is your CPU's model?

Ryzen 7 5800H

### Please describe the behaviour in detail.

The brightness control, both using Fn + brightness key and brightness slider only work when using HDR mode in Settings > Display. But when I turn it off (HDR -> SDR), the brightness aren't changing despite value of the slider are changing (slider work, but brightness still remain at 100%). This behaviour also occour on shutdown/reboot, before laptop shutdown/restart, screen goes to full brightness about 3 second. 

Tried fresh install, changing to SDR and then reboot, changing HDR to SDR didn't solve the problem. This bug happen in both macOS that I've tested: Ventura & Sonoma.

Display spec: 2560x1600 IPS 165Hz with HDR10.
NootedRed version: commit f7baa7a, but happend even in previous release commit 11f9df4.

### What should've happened instead?

Brightness control should work both in SDR and HDR mode.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

_Will attach later.._

RasyiidWho, 2024-08-13:
Tried to checkout old commit (eff598f), as in #236 just to check if the problem are presist, and it is. Even in commit eff598f, SDR brightness control are broken.

MacintoshSSD, 2024-08-16:
Seens like it happens on Raven based CPU too :/

VisualEhrmanntraut, 2024-08-16:
This seems more like a hardware limitation than a bug.

VisualEhrmanntraut, 2024-09-20:
Tracking in #293

closed 2024-09-20 by VisualEhrmanntraut