ChefKissInc/NootedRed #291: KVM Sonoma hangs at boot

Myp3a, 2024-08-23:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 5700G

### Please describe the behaviour in detail.

Hello! I'm trying to set up macOS on QEMU/KVM with GPU passthrough. Fixed most of the things, the next one is GPU.  

When I'm trying to boot with NootedRed, system just hangs at GPU initialization. Logs stop, and there's a reboot in the next few seconds (even with `keepsyms=1 -v debug=0x100`). Without the kext, system boot fine and recognizes the GPU - but no graphics acceleration available, as expected. With or without the kext, the HDMI output is blank (contains last `udev` messages before `vfio-pci` takeover).

My host OS is:
- OS: Arch Linux (linux-zen 6.10.6, ACS patch enabled)
- CPU: Ryzen 7 5700G
- RAM: 48GB
- iGPU: Vega 8 (VFIO passthrough with linked audio device)
- dGPU: RX6700XT / Tesla P100 (stays on host, macOS knows nothing about them)

My guest macOS is:
- OS: macOS Sonoma 14.6.1
- CPU: Haswell-noTSX emulation layer, 4 cores
- RAM: 16GB
- QEMU display, VNC, SSH
- Loader: OpenCore 1.0.1
- Kexts: Lilu, AppleALC, MCEReporterDisabler, NootedRed, USBPorts, VirtualSMC

My config.plist: [config.txt](https://github.com/user-attachments/files/16726714/config.txt)  
<details><summary>My QEMU boot command</summary>

```
#!/bin/bash

args=(
    -nodefaults 
    -cpu Haswell-noTSX,vendor=GenuineIntel,+hypervisor,+invtsc,vmware-cpuid-freq=on,+ssse3,+sse4.2,+popcnt,+avx,+aes,+xsave,+xsaveopt,check 
    -smp 4,sockets=1,dies=1,cores=4,threads=1 
    -m 16G 
    -machine type=q35,smm=off,graphics=off,vmport=off,dump-guest-core=off,hpet=off,accel=kvm 
    -enable-kvm 
    -global kvm-pit.lost_tick_policy=discard 
    -display none
    -vga vmware 
    -monitor stdio 
    -name macos,process=macos,debug-threads=on 
    -device nec-usb-xhci,id=xhci 
    -device usb-kbd,bus=xhci.0 
    -global nec-usb-xhci.msi=off 
    -device usb-tablet 
    -netdev user,id=net0,hostfwd=tcp::2222-:22 -device vmxnet3,netdev=net0,id=net0,mac=52:54:00:c9:18:27
    -device virtio-blk-pci,drive=InstallMedia,scsi=off,bus=pcie.0,addr=0x6 
    -drive file=/home/ivga/macos/system/base.dmg,id=InstallMedia,format=dmg,cache=unsafe,readonly=on,if=none 
    -drive file=/home/ivga/macos/system/data.img,id=data3,format=raw,cache=none,aio=native,discard=on,detect-zeroes=on,if=none 
    -device virtio-blk-pci,drive=data3,scsi=off,bus=pcie.0,addr=0xa,iothread=io2,bootindex=3 
    -object iothread,id=io2 
    -device virtio-blk-pci,drive=OpenCore,scsi=off,bus=pcie.0,addr=0x5,bootindex=9 
    -drive file=/home/ivga/macos/system/boot.img,id=OpenCore,format=raw,cache=unsafe,readonly=on,if=none 
    -smbios type=2 
    -rtc base=utc,base=localtime 
    -global ICH9-LPC.disable_s3=1 
    -global ICH9-LPC.disable_s4=1 
    -global ICH9-LPC.acpi-pci-hotplug-with-bridge-support=off 
    -device isa-applesmc,osk=ourhardworkbythesewordsguardedpleasedontsteal\(c\)AppleComputerInc 
    -drive if=pflash,format=raw,readonly=on,file=/home/ivga/macos/system/macos.rom 
    -drive if=pflash,format=raw,file=/home/ivga/macos/system/macos.vars 
    -object rng-random,id=objrng0,filename=/dev/urandom 
    -device virtio-rng-pci,rng=objrng0,id=rng0,bus=pcie.0,addr=0x1c 
    -device virtio-balloon-pci,id=balloon0,bus=pcie.0,addr=0x4 
    -device vfio-pci,host=0a:00.0
    -device vfio-pci,host=0a:00.1
    -vnc 0.0.0.0:1,password=on
)

qemu-system-x86_64 "${args[@]}"
```
</details>
<details><summary>IORegExplorer showing the passed GPU</summary>

[image]
</details>
<details><summary>GPU in system information</summary>

[image]
</details>
<details><summary>OS boot hang location</summary>

[image]
</details>

Thank you for the great work! I hope this one can be solved too. Really want the acceleration for snappy UI, and host doesn't need the iGPU at all

### What should've happened instead?

macOS should boot as expected and provide graphics acceleration

nononymousse, 2024-08-23:
might be the same as #235

Myp3a, 2024-08-23:
It might, but #235 mentions crashes, while system is properly booted. Mine, in contrast, doesn't boot at all

VisualEhrmanntraut, 2024-08-23:
> might be the same as #235

Nah. I think even Linux has trouble with AMD iGPUs being passed through to it. Or at least it did in the past. I would suggest waiting a little bit as I'm working on a semi-big bug fix and rewrite commit.

VisualEhrmanntraut, 2024-08-23:
The changes in questions have been pushed. Let's see if it is fixed.

Myp3a, 2024-08-24:
Nope, still stuck at the exact same place with reboot

VisualEhrmanntraut, 2024-08-24:
Well, when it gets stuck here, it means it can't get the VBIOS. Perhaps you need to inject your system's VFCT ACPI table into the VM somehow.

Myp3a, 2024-08-24:
The VFCT table couldn't be injected, it's too big for QEMU. vfio-pci's romfile option also doesn't work. However, I can dump the VBIOS from my host OS and get a file with it - is there a way to load VBIOS from file instead of ACPI?

VisualEhrmanntraut, 2024-08-24:
Probably should not instruct him to rebuild the entire kext. Just use OpenCore's DeviceProperties method.

Myp3a, 2024-08-24:
Wow, guys, you rock! I did some changes to PCI QEMU settings and injected the VBIOS - and it works!  
Now I've got two small problems:
 - My HDMI output (the only one) is blank
 - VM reboot throws it into bootloop (probably expected, as iGPU doesn't get proper deinitialization - full host reboot solves it)

I wonder if these are the easy fixes, but if not, I'm fine with this - fully covers my use-case

Myp3a, 2024-08-26:
My MB died, so I was forced to buy a new one, and it's doesn't play well with macOS. I get a hard host reboot (not even a kernel panic) every time I turn my VM on. This isn't a VBIOS issue, probably - I have extracted a new one. No displays are plugged into the iGPU. Once again, if this is easily fixable - I'm all ears, otherwise feel free to close the issue, because this is becoming more about my problems than the kext ones.
<details><summary>Last log lines before reboot</summary>

[image]
</details>

PRBB28, 2024-09-10:
Maybe this is a bit late, you need to remove the vmware vga when you passhtrough the igpu. I had the same problem but under proxmox, but should be similar since it uses kvm as well.

Myp3a, 2024-09-17:
But how can I debug the boot process without the virtual display?

With virtual display and physical display connected the system still reboots.
Without virtual display and with physical display it boots to some point that my router registers the DHCP lease. Afterwards, host system hangs for a while but survives that without rebooting. However, VM is stuck at 100% CPU and doesn't respond to anything (and display is blank)

VisualEhrmanntraut, 2024-09-17:
you can add a serial device and send its output to a file

Molay, 2024-10-03:
Please check whether the virtual machine has the following configuration.
1. The 5700G will have devices from 0x:00.0 to 0x:00.6, such as 04:00.0 (core graphics), 04:00.1 (core graphics audio), 04:00.2 (Platform Security Processor), 04:00.3 (USB 3.1 controller), etc.
By default, these devices share one IOMMU Group.
If you need to pass the core graphics and core graphics audio to the virtual machine, you need to set up a separate IOMMU Group for each.
For details, you can search for how to set up an IOMMU Group.
2. You need to set the correct VBIOS for the core graphics and set it as the primary GPU.
```
hostpci0: 0000:04:00.0,pcie=1,romfile=5700G.rom,x-vga=1
hostpci1: 0000:04:00.1,pcie=1,rombar=0
vga: none
```
3. The core graphics needs to be placed on the pci.0 bus.
```
args: -set device.hostpci0.bus=pci.0 -set device.hostpci0.addr=0x01.0 -set device.hostpci1.bus=pci.0 -set device.hostpci1.addr=0x02.0
```

After all these settings are completed, it should be able to start normally.

However, it should be noted that the restart of the virtual machine will be stuck because the integrated graphics hardware cannot be automatically reset. It can only be restored by restarting Proxmox VE. There may be a solution later.

Myp3a, 2024-10-03:
Yeah, I've set up my IOMMU groups with Zen kernel and ACS override patch. VBIOS isn't an issue, I'm passing it to the GPU by injecting it with device properties. GPU is also on the PCI.0 bus.

I'm a bit puzzled that it worked for a moment, but doesn't work with another MB and another (but same model) CPU. Still setting up the debug environment, though, will be back later with serial logs

Myp3a, 2024-10-03:
My boot log via serial output (set it up with the instruction [here](https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/kernel-debugging.html)). Looks like a lot is missing, am I doing it properly?
<details><summary>Boot log</summary>

```
kprintf initialized
version_variant = 0
version         = Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
KASLR slide: 0x0000000007800000 dynamic
Hiding local relocations
WARNING: ignoring first page in [0x0:0x86]
Physical memory 8192 MB
avail_remaining = 0x1ff6e6
Kernel virtual space from 0xffffff8000000000 to 0xffffffffffffefff.
Available physical space from 0xc3cf000 to 0x27ffff000
initialize_screen: b=00000000, w=00000000, h=00000000, r=00000000, d=00000001
initialize_screen: No video - forcing serial mode
IOKit IOMD setownership ENABLED
ATM subsystem is initialized
Telemetry: Sampling all tasks once per 1 second
Scheduler: Default of dualq
Setting scheduler priority decay band limit 18
BANK subsystem is initialized
IPC_PTHREAD_PRIORITY subsystem is initialized
Reallocated master cpu data: 0xffffff80088f3980, interrupt stack: 0xffffff80145e1000, fault stack: 0xffffff800790c800
CPU identification: Intel Core Processor (Haswell, no TSX)
CPU features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH MMX FXSR SSE SSE2 HTT SSE3 PCLMULQDQ SSSE3 FMA CX16 SSE4.1 SSE4.2 x2APIC MOVBE POPCNT AES VMM XSAVE OSXSAVE TSCTMR AVX1.0 RDRAND F16C RDWRFSGS BMI1 AVX2 SMEP BMI2 ERMS INVPCID
  HTT: 4 cores per package; 4 logical cpus per package
CPU extended features: SYSCALL XD EM64T LAHF LZCNT RDTSCP TSCI
Initializing EFI runtime services
MSR_IA32_APIC_BASE 0xfee00000 enabled legacy mode BSP
x2APIC supported and will be enabled
x2APIC mode enabled
ID: 0x0 LDR: 0x1
Boot cpu local APIC id 0x0
TSC sync margin disabled
fpu_state: AVX, state_size: 832
Kernel text 0xffffff8007ae4000-0xffffff80084e4000 to be write-protected
Marking (0xffffff8007a10000, 0xffffff8007ae0000) as rwx
Marking (0xffffff8008b64000, 0xffffff8008ba0000) as rwx
Marking (0xffffff8008ba0000, 0xffffff8008ba1000) as rwx
Marking (0xffffff80084e4000, 0xffffff80084f6000) as rwx
Marking (0xffffff8007ae0000, 0xffffff8007ae1000) as rwx
Marking (0xffffff8008ba1000, 0xffffff8008ba1000) as rwx
Marking (0xffffff80084f6000, 0xffffff80084f6000) as rwx
Marking (0xffffff800b930000, 0xffffff800bcbd000) as rwx
Marking (0xffffff800a8b8000, 0xffffff800a903000) as rwx
Marking (0xffffff8007a10000, 0xffffff8007a10000) as rwx
Marking (0xffffff800a94c000, 0xffffff800b453000) as rwx
[RTCLOCK] frequency 3790000000 (3792887000)
Registered coredump handler for kernel
Kernel boot args: 'keepsyms=1 -v -NRedDebug debug=0x12a msgbuf=1048576 '
corecrypto_kext_start (tracing disabled)
FIPSPOST_KEXT [358126726] fipspost_post:154: [FIPSPOST][Module-ID] Apple corecrypto Module v14.0 [Intel, Kernel, Software, SL1]
FIPSPOST_KEXT [360096273] fipspost_post:162: PASSED: (0 ms) - fipspost_post_hmac
FIPSPOST_KEXT [362037817] fipspost_post:166: PASSED: (0 ms) - fipspost_post_integrity
FIPSPOST_KEXT [363339484] fipspost_post:170: PASSED: (0 ms) - fipspost_post_indicator
FIPSPOST_KEXT [364601235] fipspost_post:171: PASSED: (0 ms) - fipspost_post_aes_ecb
FIPSPOST_KEXT [365844903] fipspost_post:172: PASSED: (0 ms) - fipspost_post_aes_cbc
FIPSPOST_KEXT [369054461] fipspost_post:173: PASSED: (1 ms) - fipspost_post_rsa_sig
FIPSPOST_KEXT [374676384] fipspost_post:174: PASSED: (4 ms) - fipspost_post_ecdsa
FIPSPOST_KEXT [376008458] fipspost_post:175: PASSED: (0 ms) - fipspost_post_ecdh
FIPSPOST_KEXT [376983233] fipspost_post:176: PASSED: (0 ms) - fipspost_post_aes_ccm
FIPSPOST_KEXT [377770316] fipspost_post:177: PASSED: (0 ms) - fipspost_post_aes_cmac
FIPSPOST_KEXT [378463373] fipspost_post:178: PASSED: (0 ms) - fipspost_post_hkdf
FIPSPOST_KEXT [384123748] fipspost_post:179: PASSED: (5 ms) - fipspost_post_pbkdf
FIPSPOST_KEXT [384878662] fipspost_post:180: PASSED: (0 ms) - fipspost_post_drbg_hmac
FIPSPOST_KEXT [385694068] fipspost_post:182: PASSED: (0 ms) - fipspost_post_kdf_ctr
FIPSPOST_KEXT [386377828] fipspost_post:183: PASSED: (0 ms) - fipspost_post_kdf_ctr_cmac
FIPSPOST_KEXT [387034617] fipspost_post:184: PASSED: (0 ms) - fipspost_post_aes_gcm
FIPSPOST_KEXT [387662072] fipspost_post:185: PASSED: (0 ms) - fipspost_post_aes_xts
[... 92 lines cut ...]
IOPPF - IODeviceTree:/efi/platform/StartupPowerEvents: 0x0
AppleIPAppender::probe(IOResources)
AppleIPAppender::start(IOResources) <1>
X86PlatformPlugin::configResourceCallback - DeepSleep is not supported!
[AGPM][INFO ][CONTROLLER  ][start                   ] Entered AGPMController::start()
[AGPM][INFO ][CONTROLLER  ][start                   ] init fSlotControl, fSlotControl = 0
[AGPM][INFO ][CONTROLLER  ][createAGPMClass         ] machines_dict->getObject(modelName iMacPro1,1)
[AGPM][INFO ][CONTROLLER  ][createAGPMClass         ] gpuDict S00
[AGPM][INFO ][CONTROLLER  ][createAGPMClass         ] build gpuDict by GPU S00.
[AGPM][WARN ][CONTROLLER  ][createAGPMClass         ] GPU = S00 unknownPlatform (no matching vendor_id+device_id or GPU name); don't create AGPM class
[AGPM][INFO ][CONTROLLER  ][start                   ] fIsExternalGPU 0
X86PlatformPlugin::cgRegisterCallbackHandler - entered, type gpu-external-plimit-notification, target AGPM
X86PlatformPlugin::cgRegisterPLimitCallbackHandler - There are 0 entries in the linked list before target name AGPM
HE2N_Key Does Not Exist, use kSMCPStatesEGPU for External GPU
[... cut ...]
[AGPM][INFO ][CONTROLLER  ][start                   ] Exited AGPMController::start()
SFA: match category IODefaultMatchCategory exists
**** [IOBluetoothHCIUserClient][SetBTRBOffWL] -- mBluetoothFamily->mACPIMethods->SetBTRB (OFF) returned 0xE00002C7 error
**** [IOBluetoothHCIUserClient][SetBTLPOffWL] -- mBluetoothFamily->mACPIMethods->SetBTLP (OFF) returned 0xE00002C7 error
**** [IOBluetoothHCIUserClient][SetBTRSWL] -- mBluetoothFamily->mACPIMethods->SetBTRS() returned 0xE00002C7 error
kPEDisableScreen 1
```

</details>

Myp3a, 2024-10-18:
Still breaks at the same place with no additional logging. I've tried the old Inject method and the new one, with `romfile=...`. Also I've rebooted the machine between retries, so they are done with the clean GPU state.

Relevant QEMU config lines:
```
-device pcie-root-port,port=0,chassis=1,id=pci.0,bus=pcie.0,multifunction=on,addr=0x1
-device vfio-pci,host=30:00.0,bus=pci.0,multifunction=on,addr=0x0,romfile=/my_macos_path/igpu_vbios_msi.rom
-device vfio-pci,host=30:00.1,bus=pci.0,addr=0x0.0x1
```

VisualEhrmanntraut, 2024-10-19:
@Myp3a Try adding `serial=3` to the boot args

Myp3a, 2024-10-19:
Done, now it looks like proper log. Attaching it to the message below.
<details><summary>Boot log</summary>

```
kprintf initialized
Serial mode specified: 00000003
version_variant = 0
version         = Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
KASLR slide: 0x000000000d200000 dynamic
Hiding local relocations
WARNING: ignoring first page in [0x0:0x86]
Physical memory 8192 MB
avail_remaining = 0x1ff6e6
PMAP: Supervisor Mode Execute Protection enabled
Kernel virtual space from 0xffffff8000000000 to 0xffffffffffffefff.
Available physical space from 0x11dc5000 to 0x27ffff000
initialize_screen: b=00000000, w=00000000, h=00000000, r=00000000, d=00000001
initialize_screen: No video - forcing serial mode
Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
pmap_startup() delaying init/free of page nums > 0x200000
pmap_startup() init/release time: 24492 microsec
pmap_startup() delayed init/release of 565595 pages
vm_page_bootstrap: 1482313 free pages, 598455 wired pages, (up to 565595 of which are delayed free)
VM boostrap: 73 maps, 153 entries and 128 holes available
Maximum number of VM swap files: 100
kext submap [0x<ptr> - 0x<ptr>], kernel text [0x<ptr> - 0x<ptr>]
"vm_compressor_mode" is 4
VM bootstrap done: 57 maps, 110 entries and 118 holes left
IOKit IOMD setownership ENABLED
ATM subsystem is initialized
OSLog stream configured: stream: 8192 bytes, cache: 2112 bytes
Firehose configured: 16 chunks, 8 io pages
Log queues configured: slot count: 32, per-slot size: 32768, total size: 1048576
Long logs support configured: size: 16384
Telemetry: Sampling all tasks once per 1 second
Scheduler: Default of dualq
Setting scheduler priority decay band limit 18
standard timeslicing quantum is 10000 us
standard background quantum is 2500 us
mig_table_max_displ = 54 mach_kobj_count = 375
BANK subsystem is initialized
IPC_PTHREAD_PRIORITY subsystem is initialized
Reallocated master cpu data: 0xffffff800e2f3980, interrupt stack: 0xffffffb18f3f0000, fault stack: 0xffffff800d30c800
CPU identification: Intel Core Processor (Haswell, no TSX)
CPU features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH MMX FXSR SSE SSE2 HTT SSE3 PCLMULQDQ SSSE3 FMA CX16 SSE4.1 SSE4.2 x2APIC MOVBE POPCNT AES VMM XSAVE OSXSAVE TSCTMR AVX1.0 RDRAND F16C RDWRFSGS BMI1 AVX2 SMEP BMI2 ERMS INVPCID
  HTT: 4 cores per package; 4 logical cpus per package
CPU extended features: SYSCALL XD EM64T LAHF LZCNT RDTSCP TSCI
Initializing EFI runtime services
MSR_IA32_APIC_BASE 0xfee00000 enabled legacy mode BSP
x2APIC supported and will be enabled
x2APIC mode enabled
ID: 0x0 LDR: 0x1
Boot cpu local APIC id 0x0
TSC sync margin disabled
fpu_state: AVX, state_size: 832
Kernel text 0xffffff800d4e4000-0xffffff800dee4000 to be write-protected
Marking (0xffffff800d410000, 0xffffff800d4e0000) as rwx
Marking (0xffffff800e564000, 0xffffff800e5a0000) as rwx
Marking (0xffffff800e5a0000, 0xffffff800e5a1000) as rwx
Marking (0xffffff800dee4000, 0xffffff800def6000) as rwx
Marking (0xffffff800d4e0000, 0xffffff800d4e1000) as rwx
Marking (0xffffff800e5a1000, 0xffffff800e5a1000) as rwx
Marking (0xffffff800def6000, 0xffffff800def6000) as rwx
Marking (0xffffff8011333000, 0xffffff80116c0000) as rwx
Marking (0xffffff80102b8000, 0xffffff8010303000) as rwx
Marking (0xffffff800d410000, 0xffffff800d410000) as rwx
Marking (0xffffff801034c000, 0xffffff8010e53000) as rwx
[RTCLOCK] frequency 3790000000 (3792888000)
TSC Deadline Timer supported and enabled
kdp_core zlib memory 0x7000
[... 1108 lines cut ...]
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002c2, -1)
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002f0, -1)
AppleKeyStore:7690:167: unexpected session uid: -1 requested by: bluetoothd (167)
AppleKeyStore:12195:167: operation failed (sel: 7 ret: e00002c2, -1)
AppleKeyStore:12195:167: operation failed (sel: 7 ret: e00002f0, -1)
IOConsoleUsers: time(0) 0->0, lin 0, llk 1,
IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sm 0x0
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
IOConsoleUsers: time(0) 0->0, lin 0, llk 1,
IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sm 0x0
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
AppleKeyStore:12195:106: operation failed (sel: 35 ret: e00002f0, -1)
AppleKeyStore:7690:171: unexpected session uid: -1 requested by: AirPlayXPCHelper (171)
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002c2, -1)
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002f0, -1)
IOConsoleUsers: time(0) 0->0, lin 0, llk 1,
IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sm 0x0
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
S00-0x195-A: newUserClient failed: 0 0xe00002be 0x1000002a1 IONDRVFramebuffer
tx_flush:1338: disk3 xid 239337 tx stats: # 40 owait 12 66241us finish 40 bar2 40 f 9 enter 62 fq 45 62 68169us close 2us prep 129us flush 109002us
```

</details>

VisualEhrmanntraut, 2024-10-19:
There's no logs about patches being done at all. Is Lilu even working? 😄

Myp3a, 2024-10-19:
I'm pretty sure that it should work: it's defined in OpenCore and located in Kexts directory. However, I've updated it to the latest 1.6.9-debug and enabled logging.
<details><summary>Logs w/ Lilu</summary>

```
kprintf initialized
Serial mode specified: 00000003
version_variant = 0
version         = Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
KASLR slide: 0x0000000009400000 dynamic
Hiding local relocations
WARNING: ignoring first page in [0x0:0x86]
Physical memory 8192 MB
avail_remaining = 0x1ff6ea
PMAP: Supervisor Mode Execute Protection enabled
Kernel virtual space from 0xffffff8000000000 to 0xffffffffffffefff.
Available physical space from 0xe020000 to 0x27ffff000
initialize_screen: b=00000000, w=00000000, h=00000000, r=00000000, d=00000001
initialize_screen: No video - forcing serial mode
Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
pmap_startup() delaying init/free of page nums > 0x200000
pmap_startup() init/release time: 25078 microsec
pmap_startup() delayed init/release of 549818 pages
vm_page_bootstrap: 1498094 free pages, 582674 wired pages, (up to 549818 of which are delayed free)
VM boostrap: 73 maps, 153 entries and 128 holes available
Maximum number of VM swap files: 100
kext submap [0x<ptr> - 0x<ptr>], kernel text [0x<ptr> - 0x<ptr>]
"vm_compressor_mode" is 4
VM bootstrap done: 57 maps, 110 entries and 119 holes left
IOKit IOMD setownership ENABLED
ATM subsystem is initialized
OSLog stream configured: stream: 8192 bytes, cache: 2112 bytes
Firehose configured: 16 chunks, 8 io pages
Log queues configured: slot count: 32, per-slot size: 32768, total size: 1048576
Long logs support configured: size: 16384
Telemetry: Sampling all tasks once per 1 second
Scheduler: Default of dualq
Setting scheduler priority decay band limit 18
standard timeslicing quantum is 10000 us
standard background quantum is 2500 us
mig_table_max_displ = 54 mach_kobj_count = 375
BANK subsystem is initialized
IPC_PTHREAD_PRIORITY subsystem is initialized
Reallocated master cpu data: 0xffffff800a4f3980, interrupt stack: 0xfffffff4f9fde000, fault stack: 0xffffff800950c800
CPU identification: Intel Core Processor (Haswell, no TSX)
CPU features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH MMX FXSR SSE SSE2 HTT SSE3 PCLMULQDQ SSSE3 FMA CX16 SSE4.1 SSE4.2 x2APIC MOVBE POPCNT AES VMM XSAVE OSXSAVE TSCTMR AVX1.0 RDRAND F16C RDWRFSGS BMI1 AVX2 SMEP BMI2 ERMS INVPCID
  HTT: 4 cores per package; 4 logical cpus per package
CPU extended features: SYSCALL XD EM64T LAHF LZCNT RDTSCP TSCI
Initializing EFI runtime services
MSR_IA32_APIC_BASE 0xfee00000 enabled legacy mode BSP
x2APIC supported and will be enabled
x2APIC mode enabled
ID: 0x0 LDR: 0x1
Boot cpu local APIC id 0x0
TSC sync margin disabled
fpu_state: AVX, state_size: 832
Kernel text 0xffffff80096e4000-0xffffff800a0e4000 to be write-protected
Marking (0xffffff8009610000, 0xffffff80096e0000) as rwx
Marking (0xffffff800a764000, 0xffffff800a7a0000) as rwx
Marking (0xffffff800a7a0000, 0xffffff800a7a1000) as rwx
Marking (0xffffff800a0e4000, 0xffffff800a0f6000) as rwx
Marking (0xffffff80096e0000, 0xffffff80096e1000) as rwx
Marking (0xffffff800a7a1000, 0xffffff800a7a1000) as rwx
Marking (0xffffff800a0f6000, 0xffffff800a0f6000) as rwx
Marking (0xffffff800d58e000, 0xffffff800d91b000) as rwx
Marking (0xffffff800c4b8000, 0xffffff800c503000) as rwx
Marking (0xffffff8009610000, 0xffffff8009610000) as rwx
Marking (0xffffff800c54c000, 0xffffff800d053000) as rwx
[RTCLOCK] frequency 3790000000 (3792872000)
TSC Deadline Timer supported and enabled
kdp_core zlib memory 0x7000
[... 1096 lines cut ...]
tx_flush:1338: disk3 xid 239500 tx stats: # 40 owait 20 170446us finish 42 bar2 40 f 10 enter 216 fq 63 71 318us close 277881us prep 313us flush 904744us
handle_mount:817: disk3s5 vol-uuid: B0679A4F-0859-427D-8BD8-261FCF3901CE block size: 4096 block count: 67057654 (unencrypted; flags: 0x201; features: 21.0.12)
handle_mount:830: disk3s5 setting dev block size to 4096 from 512
apfs_log_op_with_proc:3091: disk3s5 mount-complete volume macos, requested by: mount_apfs (pid 372); parent: com.apple.Mobile (pid 342)
ioqueue_depth = 256,   ioscale = 8
apfs_log_op_with_proc:3091: disk3s5 unmounting volume macos, requested by: com.apple.MobileSoftwareUpdate. (pid 342); parent: launchd (pid 1)
apfs_vfsop_unmount:3221: disk3s5 waiting for purgatory cleaner to finish
AppleKeyStore:7690:171: unexpected session uid: -1 requested by: AirPlayXPCHelper (171)
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002c2, -1)
AppleKeyStore:12195:171: operation failed (sel: 7 ret: e00002f0, -1)
authapfs_seal_restore:598: disk3s5 integrity protection will be re-enabled
apfs_vfsop_unmount:3567: disk3 nx_num_vols_mounted is 5
apfs: total mem allocated: 69243739 (66 mb);
apfs_vfsop_unmount:3580: all done.  going home.  (numMountedAPFSVolumes 5)
IOConsoleUsers: time(0) 0->0, lin 0, llk 1,
IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sm 0x0
Lilu    config: @ (DBG) PE_initialize_console 5
S00-0x1a5-A: newUserClient failed: 0 0xe00002be 0x1000002b9 IONDRVFramebuffer
Lilu    config: @ (DBG) PE_initialize_console 5
S00-0x1a5-A: newUserClient failed: 0 0xe00002be 0x1000002b9 IONDRVFramebuffer
Couldn't build index for com.apple.kextstat: Not eligible for acceleration
tx_flush:1338: disk3 xid 239520 tx stats: # 60 owait 23 157082us finish 63 bar2 60 f 12 enter 79 fq 38 48 328us close 28us prep 174us flush 139461us
```

</details>

VisualEhrmanntraut, 2024-10-19:
That’s a bit better, still not useful though. I’m not sure what’s going on here, I’d suggest reconfiguring the EFI as per https://chefkissinc.github.io/guides/hackintosh and seeing if the behaviour changes afterwards.

Myp3a, 2024-10-19:
I think I have a clue: my log starts with `initialize_screen: No video - forcing serial mode`. So macOS thinks there's no display, and because of that GPU isn't initialized at all.

But if I'm adding a QEMU VGA display, everything starts to work - NootedRed loads and patches proper values. However, system is thrown into a reboot afterwards, so we're back at square one. It recognizes the videocard, gets VBIOS and probably behaves as expected. VBIOS is probably fine - if I try other's people VBIOS, system breaks even earlier.

<details><summary>My boot log with VGA attached</summary>

```
kprintf initialized

Serial mode specified: 00000003

version_variant = 0

version         = Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64

KASLR slide: 0x0000000003200000 dynamic

Hiding local relocations

WARNING: ignoring first page in [0x0:0x0]

Physical memory 8192 MB

avail_remaining = 0x1ff6e7

PMAP: Supervisor Mode Execute Protection enabled
Kernel virtual space from 0xffffff8000000000 to 0xffffffffffffefff.

Available physical space from 0x7e20000 to 0x27ffff000

initialize_screen: b=80000000, w=00000780, h=00000438, r=00001E00, d=00000001

Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:00 PDT 2024; root:xnu-10063.141.2~1/RELEASE_X86_64
pmap_startup() delaying init/free of page nums > 0x200000
pmap_startup() init/release time: 25046 microsec
pmap_startup() delayed init/release of 524727 pages
vm_page_bootstrap: 1523182 free pages, 557586 wired pages, (up to 524727 of which are delayed free)
VM boostrap: 73 maps, 153 entries and 128 holes available
Maximum number of VM swap files: 100
kext submap [0x<ptr> - 0x<ptr>], kernel text [0x<ptr> - 0x<ptr>]
"vm_compressor_mode" is 4
VM bootstrap done: 57 maps, 110 entries and 118 holes left
IOKit IOMD setownership ENABLED

ATM subsystem is initialized

OSLog stream configured: stream: 8192 bytes, cache: 2112 bytes
Firehose configured: 16 chunks, 8 io pages
Log queues configured: slot count: 8, per-slot size: 32768, total size: 262144
Long logs support configured: size: 16384
Telemetry: Sampling all tasks once per 1 second

Scheduler: Default of dualq

Setting scheduler priority decay band limit 18

standard timeslicing quantum is 10000 us
standard background quantum is 2500 us
mig_table_max_displ = 54 mach_kobj_count = 375
BANK subsystem is initialized

IPC_PTHREAD_PRIORITY subsystem is initialized

Reallocated master cpu data: 0xffffff80042f3980, interrupt stack: 0xffffffe228b26000, fault stack: 0xffffff800330c800

CPU identification: Intel Core Processor (Haswell, no TSX)

CPU features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH MMX FXSR SSE SSE2 SSE3 PCLMULQDQ SSSE3 FMA CX16 SSE4.1 SSE4.2 x2APIC MOVBE POPCNT AES VMM XSAVE OSXSAVE TSCTMR AVX1.0 RDRAND F16C RDWRFSGS BMI1 AVX2 SMEP BMI2 ERMS INVPCID

CPU extended features: SYSCALL XD EM64T LAHF LZCNT RDTSCP TSCI
[... 1980 lines cut ...]
initialize normal
AMD Error: Invalid size or CPU virtual address size = 0x0000000000000000, memblock_ptr = 0

AMD TTL AMDLOG: [2:0:0] PSP: event_id=0xc0201 event_info:type=2 hw_id=0 event_specific_tag=0xc pData=0 data_size=0 psp_sw_init finished 

AMD Error: [2:0:0] Error SW_IP_CLIENT_ID__GC: event_id=0xc00c0401 event_info:type=3 hw_id=0 event_specific_tag=0x0 pData=0 data_size=0 

AMD Error: Firmware atidmcub_0.dat not found in directory, for deviceId 0x00001638

AMD Error: Unable to get firmware atidmcub_0.dat

AMD TTL COS:  [2:0:0]: Write PCI Config offset 0x60 size 2 data 0x100

AMD TTL COS:  [2:0:1]: Write PCI Config offset 0x60 size 2 data 0x100

AMD TTL AMDLOG: [2:0:0] PSP: event_id=0x100211 event_info:type=2 hw_id=0 event_specific_tag=0x0 pData=0 data_size=0 psp_hardware_initialization began loading PSP FWs 

AMD TTL AMDLOG: [2:0:0] PSP: event_id=0x100214 event_info:type=2 hw_id=0 event_specific_tag=0x0 pData=0 data_size=0 psp_hardware_initialization psp_bootloader_load_sysdrv begin 

AMD TTL AMDLOG: [2:0:0] PSP: event_id=0x100214 event_info:type=2 hw_id=0 event_specific_tag=0x0 pData=0 data_size=0 psp_hardware_initialization psp_bootloader_load_sysdrv end 

AMD TTL AMDLOG: [2:0:0] PSP: event_id=0x100215 event_info:type=2 hw_id=0 event_specific_tag=0x0 pData=0 data_size=0 psp_hardware_initialization psp_bootloader_load_sos (tos) begin 

AMD TTL AMDLOG: [2:0:0] PSP: event
```

VisualEhrmanntraut, 2024-10-20:
Honestly it seems like a Lilu issue to me. I think it only starts patching when the display is initialised

PRBB28, 2024-10-20:
@Myp3a., Have you tried doing igpu passthrough with windows vm? Just to verify that you igpu passthrough is working properly.  I had hard time making it work in proxmox, and that was with windows vm. Seen lots of discussion about 5700G igpu passhtorugh not working.

Myp3a, 2024-10-20:
@PRBB28, it worked with macOS for a bit - then I was forced to change MB and CPU (same CPU, different MB), and now it doesn't work again, so 5700G passthrough is definitely possible. However, I'll check same configuration with Win/Linux, it might bring more clues to the scene.

@VisualEhrmanntraut, ty, I'll try opening an issue in Lilu bugtracker. Will come back here later with information from them

VisualEhrmanntraut, 2024-10-23:
Well, as I see it's a problem with Lilu, so I can't really do anything about this. I guess you'll need to find a way to get GOP working with a passed-through GPU.

closed 2024-10-23 by VisualEhrmanntraut