ChefKissInc/NootedRed #293: 未来的Hackintosh是否会支持相对完美的兼容性的OLED屏幕？

if-seventten, 2024-08-30:
### macOS Version

None

### What is your CPU's model?

AMD 5800小时

### Please describe the feature in detail.

I apologize for my limited English skills. Currently, the computer I'm using is equipped with an OLED display, but in the Hackintosh system, I'm unable to access its maximum brightness level or utilize the brightness adjustment function. My own technical knowledge in this area is quite limited. Regardless of whether this issue can be resolved, I sincerely look forward to your reply.

### What alternatives have you tried, if any?

Currently, I'm using software to control the brightness, but I'm still unable to access the maximum brightness level of the screen.

VisualEhrmanntraut, 2024-08-30:
enable hdr

VisualEhrmanntraut, 2024-08-30:
note: this will likely cause a link training issue, causing the display backlight to rapidly flash on boot and won't get past it

if-seventten, 2024-08-30:
Yes, I encountered the issue you mentioned after enabling HDR. Is this problem something that can be resolved?

VisualEhrmanntraut, 2024-11-05:
This behaviour is likely intentional, so closing. #301 is the real issue to fix.

closed 2024-11-05 by VisualEhrmanntraut