ChefKissInc/NootedRed #296: After 10 seconds of entering the system, the screen will restart

Win10Q, 2024-09-07:
### macOS Version

None

### What is your CPU's model?

AMD RYZEN 3500u

### Please describe the behaviour in detail.

After 10 seconds of entering the system, the screen will restart，i used the latest nootedred
[image]
[image]

### What should've happened instead?

don't reboot

treydb, 2024-09-07:
You’ll want to use WhateverGreen instead of Nooted so that you can change your wallpaper to a static image. Also, you want to make sure your iCloud photo - if you’re signed in - is a static, regular image. After making these changes, swap back to Nooted.

The troubleshooting information suggests making these changes and it helped me on my 3550H.

VisualEhrmanntraut, 2024-09-16:
Assuming issue is resolved as this is the common working workaround. If not, please make a comment. This issue is likely #235