ChefKissInc/NootedRed #298: Random HWLibs failures in macOS Ventura

Zormeister, 2024-09-09:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 5 5625U

### Please describe the behaviour in detail.

Boils down to:
Early init in the accel stack fails, TTL::initialise returns `kIOReturnError`

### What should've happened instead?

Should be booting with full acceleration, instead when it happens it reverts to the Framebuffer.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[dm.txt](https://github.com/user-attachments/files/16928344/dm.txt)
[dm2.txt](https://github.com/user-attachments/files/16928345/dm2.txt)

VisualEhrmanntraut, 2024-09-16:
We cannot replicate this issue, therefore further information will be required for resolution. I will update you further in the coming days with further instructions.

VisualEhrmanntraut, 2024-09-21:
With the latest commit if you have `-NRedDebug` in your boot-args you will get a shit ton of logs in dmesg. Using https://github.com/ChefKissInc/DiskDbg you will be able to have it saved in `/Users` and send them here for further inspection. You will need to use https://github.com/acidanthera/DebugEnhancer as per usual as the dmesg will be several MBs. You know how to do this, so I hope you don't ask me to hold your hand through the process again.

VisualEhrmanntraut, 2024-09-27:
Timed out waiting for response.

closed 2024-09-27 by VisualEhrmanntraut