ChefKissInc/NootedRed #300: HDMI audio output can't sound out when unplugged and reinserted

Anonyame, 2024-09-17:
### macOS Version

Big Sur

### What is your CPU's model?

AMD Ryzen 7 5800H with Radeon Graphics

### Please describe the behaviour in detail.

In my workflow, I use a VX2731-2K-HD monitor (with 2 HDMI + 1 DP + 1 x 3.5mm audio output) connected to a Beelink SER5 Max running Hackintosh with the latest version of NRed and another desktop PC with Windows installed. The Beelink SER5 Max is connected to VX2731-2K-HD via HDMI, while the desktop PC connects through DP.

Upon startup, I switch VX2731-2K-HD's input source to HDMI for audio output which works fine; then when switching to the desktop with Windows installed and using it as an input source, its audio can properly pass through VX2731-2K-HD. However, after returning the monitor's input source to Beelink SER5 Max, no sound is emitted from 3.5mm Audio Output on VX2731-2K-HD anymore.

If I switch back to using the desktop PC as an input source again at this point, then indeed audio can still be outputted through the 3.5mm Audio Output of VX2731-2K-HD.

[image]
[image]
[image]

### What should've happened instead?

VX2731-2K-HD should be able to continue outputting the sound from the Beelink SER5 Max Hackintosh via its 3.5mm audio output.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

No crash log created during this action.

Anonyame, 2024-09-17:
I am preparing to test if, when I set the input source of VX2731-2K-HD to DP during the Beelink SER5 Max startup and then switch it back to HDMI after booting up, the output will be normal or not.

Anonyame, 2024-09-17:
> I am preparing to test if, when I set the input source of VX2731-2K-HD to DP during the Beelink SER5 Max startup and then switch it back to HDMI after booting up, the output will be normal or not.

Tested. Can't work.

Anonyame, 2024-09-20:
Test for more times, and find that this phenomenon sometimes occurs and sometimes it doesn't. Currently I'm preparing for more observations to determine whether this issue is just a rare occurrence or not.

Thx!

VisualEhrmanntraut, 2024-09-20:
I was able to replicate something similar by accident actually. After reconnecting the cable the sound device didn't show up, and if I reconnect the cable again it shows up.

Anonyame, 2024-09-20:
> I was able to replicate something similar by accident actually. After reconnecting the cable the sound device didn't show up, and then I reconnected the cable again and it showed up.

Thx for your test! The phenomenon here is pretty much the same as what you have there, however, when I first encountered this issue, no matter how many times I switched or unplugged and reconnected the HDMI cable, the sound device did not show up in the system info list. But during my latest test, despite multiple attempts at switching devices around, the sound device connected properly each time. Really very weird.

Anonyame, 2024-09-23:
Test for more times and a temporary solution that doesn't require modifying the source code. Follow these steps:

1. Lock the screen and wait for a while until macOS shut the screen down;
2. Wait for a while, left the screen off;
3. Wake macOS up, the screen will open, unlock the system.
4. And we will find that the sound device has been back.

Idk whether I need to close the issue or not, should I close it?

Thx!

VisualEhrmanntraut, 2024-11-05:
Merging with #225 as it's probably the same underlying cause.