ChefKissInc/NootedRed #304: Anydesk Crash When i Try to access another Computer

taqieallah, 2024-10-02:
### macOS Version

Sequoia

### What is your CPU's model?

Ryzen 5 5650u

### Please describe the behaviour in detail.

ive my ho 845 g8 with ryzen 5 5650u .. 512 mb vram and when i try to acceess anydesk on any other pc the app crash without any error msg but when some try to access my pc it works good .. any help on this?

### What should've happened instead?

it should work good

alvindimas05, 2024-10-02:
@taqieallah AnyDesk works on my machine which has R5 5600H and 1 GB VRAM. Please increase your VRAM and if you can't, then you have no luck. I'm not sure if you can disable hardware acceleration on AnyDesk.

VisualEhrmanntraut, 2024-10-20:
Closing due to inactivity

closed 2024-10-20 by VisualEhrmanntraut