ChefKissInc/NootedRed #307: Weird Third-Party softwares startup

anHiep, 2024-10-12:
### macOS Version

Ventura

### What is your CPU's model?

AMD ryzen 5 5500u

### Please describe the behaviour in detail.

When I clicked on the third-party apps like discord, edge, notion, .... Everything in my screen freeze and I can't do anything. After about 1 minutes, everything turn back working properly. I fix this for edge and chrome by disable GPU rasterization and vscode by disable GPU acceleration. But some apps like notion cannot be changed like that.

### What should've happened instead?

After researching, I think that the problem can be solved by increasing the UMA buffer sizes

VisualEhrmanntraut, 2024-10-12:
#158

closed 2024-10-12 by VisualEhrmanntraut

anHiep, 2024-10-12:
But I can't find out how to fix for apps like vscode and notion

VisualEhrmanntraut, 2024-10-12:
Creating a new issue won't magically make a fix appear. As you pointed out in your issue description, using UMAF to increase the VRAM size should help. If you can't, there is nothing else to do.