ChefKissInc/NootedRed #312: GPU waking up my hackintosh cause extreme battery drain!

ShashwanthN, 2024-10-18:
### macOS Version

Sonoma

### What is your CPU's model?

AMD RYZEN 5 5500U

### Please describe the behaviour in detail.

When I put the computer to sleep it keeps waking up within 2 seconds.
I still hear whatapp notification in sleep, battery drain is quite insane, getting about 2 hours of battery life. the fans keep running hard.

here are my logs, not sure what to do!

`	Entering DarkWake state due to 'Software Sleep pid=165':TCPKeepAlive=disabled Using Batt (Charge:53%) 86 secs   
2024-10-17 22:17:22 +0530 Wake                	DarkWake to FullWake from Invalid [CDNVA] : due to UserActivity Assertion Using BATT (Charge:52%) 11 secs   
2024-10-17 22:17:33 +0530 Sleep               	Entering DarkWake state due to 'Software Sleep pid=165':TCPKeepAlive=disabled Using Batt (Charge:52%) 6833 secs 
2024-10-18 00:11:26 +0530 Wake                	DarkWake to FullWake from Invalid [CDNVA] : due to UserActivity Assertion Using BATT (Charge:25%)           
what does this the invalide cdnva mean search the entire internet for similar issue`

### What should've happened instead?

should have gone to sleep. and preserved battery life.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

I couldnt find file and yes I have keepsyms=1 enabled.

VisualEhrmanntraut, 2024-10-20:
I don't think this is a NootedRed issue, I think you just haven't set up sleep properly.

ShashwanthN, 2024-10-20:
Really? Ive crossed out stuff that would break sleep. for a month ive been trylessly trying, In the end the last thing that couldve caused this was GPU. I suppose You will know the best haha Thanks for you reply!

closed 2024-10-20 by ShashwanthN