ChefKissInc/NootedRed #316: Failed to boot into MacOS Sequoia (15.1)

melamit, 2024-10-30:
### macOS Version

Sequoia

### What is your CPU's model?

AMD Ryzen 5 3400G (Radeon Vega 11)

### Please describe the behaviour in detail.

MacOS fails to boot everytime i use NootedRed. In the logs i have seen something like Lilu, failed to resolve and some numbers, honestly idk if it's related

### What should've happened instead?

MacOS should have been booted sucessfully as it did a few timesbefore

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

I weren't able to boot into MacOS properly, are the old logs kept in /Library/Logs/DiagnosticReports?

VisualEhrmanntraut, 2024-10-30:
NootedRed has been reported to work on macOS 15.1. This isn't a support forum.

melamit, 2024-10-30:
I am very sorry for this, but where else i should report this? I tried a few _solutions_ from other issues but they didn't worked

AlfCraft07, 2024-10-30:
It works, it's skill issue. Try posting on r/hackintosh.

VisualEhrmanntraut, 2024-10-30:
Please refrain from using this kind of phrasing, it's uncalled for.

AlfCraft07, 2024-10-30:
> Please refrain from using this kind of phrasing, it's uncalled for.

Sorry. Didn't think I could be too harsh.

melamit, 2024-10-30:
Closing the issues, moving up to the support forums

VisualEhrmanntraut, 2024-10-30:
Sorry for my colleagues phrasing, I'm sure he didn't mean to insult you. One of my acquaintances told me he had the same issue and he fixed it by updating NootedRed.

VisualEhrmanntraut, 2024-10-30:
(I'd like to note here that the Lilu errors are a non-issue)