ChefKissInc/NootedRed #322: Merge NootRX into NootedRed

VisualEhrmanntraut, 2024-11-05:
### macOS Version

None

### What is your CPU's model?

N/A

### Please describe the feature in detail.

N/A

### What alternatives have you tried, if any?

N/A

VisualEhrmanntraut, 2024-11-05:
[image]

VisualEhrmanntraut, 2025-09-29:
[image]

keybordguy, 2025-11-08:
Any updates on this and adding rdna 2 igpu support since i couldnt comment on there?

keybordguy, 2026-03-14:
> Any news..!? This feature would be the top of the top 😍

i had the same question for 4 months and my guess is that the developers of this kext are trying to make it PERFECT instead of adding a bunch of features and all of them are buggy.

VisualEhrmanntraut, 2026-05-10:
The removal of the model mechanism in the latest release is part of this work btw.

HungLe17, 2026-10-03:
any updates on rdna igpu?