ChefKissInc/NootedRed #324: After the update on November 8, 2024, the model of the graphic adapter is incorrectly detected ( Ryzen 5 4600H)

PhilVlady, 2024-11-08:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 4600H

### Please describe the behaviour in detail.

After the update on November 8, 2024, the model of the graphic adapter is incorrectly detected ( Ryzen 5 4600H).
It is now determined not as Vega 6, but as AMD Radeon RX.

[image]
[image]

### What should've happened instead?

Should be defined as Vega.

VisualEhrmanntraut, 2024-11-08:
Not a bug; these models are extracted directly from the Windows driver. Complain to AMD.

closed 2024-11-08 by VisualEhrmanntraut

PhilVlady, 2024-11-08:
But until today's update there was no display "Radeon RX".
This is a screenshot from a few hours ago.
[image]

VisualEhrmanntraut, 2024-11-08:
Radeon RX is correct. I just made it consistent, after I updated the models.
[image]

PhilVlady, 2024-11-08:
Do I need to change some values ​​to get the correct look or just leave it as is? Thanks!

VisualEhrmanntraut, 2024-11-08:
"AMD Radeon RX Graphics" just means there is no specific branding name for your card. No additional configuration is required.