ChefKissInc/NootedRed #325: The "Chess" application does not start.

PhilVlady, 2024-11-08:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 5 4600H

### Please describe the behaviour in detail.

The "Chess" application does not start. An error window immediately pops up. Although acceleration is enabled and works, as far as I can see.
[image]

### What should've happened instead?

Srart the "Chess" application.

VisualEhrmanntraut, 2024-11-08:
Please stop making unnecessary bug reports. As stated in the website, you need to increase your VRAM size to 1GB or more.

closed 2024-11-08 by VisualEhrmanntraut

PhilVlady, 2024-11-08:
The processor has 512 MB of video memory allocated at the physical level. This value cannot be increased. I understand you. Thank you!

VisualEhrmanntraut, 2024-11-08:
If you haven't tried Smokeless UMAF, you can try increasing the size using that. If it doesn't work, then you're out of luck for now, unless I find a way to let macOS dynamically allocate the memory.

PhilVlady, 2024-11-08:
Ooooh!!! You are a great Guru! I didn't know about the "Smokeless UMAF" program! Now everything is great! I didn't expect such fast and qualified help! My gratitude! 
[image]
[image]

VisualEhrmanntraut, 2024-11-08:
@PhilVlady You’re welcome, feel free to join the Telegram linked in our website or create a new post in the discussions tab for further questions regarding NootedRed.