ChefKissInc/NootedRed #331: DMCUB service patch leads to crash in macOS Catalina

Zormeister, 2024-11-19:
### macOS Version

Catalina

### What is your CPU's model?

AMD Ryzen 5 5625U

### Please describe the behaviour in detail.

Used the latest NRed CI, alongside the initial DMCUB patch build.

This resulted in a crash.

Sorry for not having reported this earlier, I didn't want to flood the GitHub issues too much, hopefully this'll be the last FB-breaking issue.

### What should've happened instead?

It should've booted to LoginWindow

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[Kernel-2024-11-19-174451.txt](https://github.com/user-attachments/files/17810793/Kernel-2024-11-19-174451.txt)

VisualEhrmanntraut, 2024-11-19:
> Sorry for not having reported this earlier, I didn't want to flood the GitHub issues too much, hopefully this'll be the last FB-breaking issue.

Unfortunately I don't have a Renoir-based device so I can't catch the issues myself. The QC testers don't seem to have Catalina installed. Maybe you're the only one actually using Catalina, at least on a Renoir-based device.

VisualEhrmanntraut, 2024-11-19:
@Zormeister That log you've shared says "macOS 13.6.9 (22G830)", not macOS 10.15.

VisualEhrmanntraut, 2024-11-19:
Oh, but the XNU version in the panic text is 19.6.0, which is Catalina. Confusing.

VisualEhrmanntraut, 2024-11-19:
Anyways
Error code: 0x2
CR2 = 0xc0800
-> write to not present address 0xc0800
and that log says __ZN14AmdDalServices10initializeEv + 0xc3
But none of those instructions are writes
[image]

Zormeister, 2024-11-19:
@VisualEhrmanntraut

Did some manual debugging, Patch 1 seems to be borked on Catalina for whatever reason.

I disabled Patch 1, which resulted in booting to LoginWindow and being able to use the OS.
As a sanity check, I re-enabled Patch 1 and disabled Patch 2, leading to the same crash in the panic log.

VisualEhrmanntraut, 2024-11-19:
No shit, but that patch is just replacing `cmp dword ptr [rcx + 0x2c], 0x8f` with `cmp eax, eax` followed by a bunch of NOPs.

VisualEhrmanntraut, 2024-11-19:
@Zormeister Does this work?

[NootedRed-1.0.0-RESEARCH_RELEASE.zip](https://github.com/user-attachments/files/17821635/NootedRed-1.0.0-RESEARCH_RELEASE.zip)

VisualEhrmanntraut, 2024-11-19:
Added additional patch, to disable DMCUB firmware loading from DAL. The default is off in Big Sur and newer, but on in Catalina, for some reason.

Zormeister, 2024-11-19:
aight, can test soon

Zormeister, 2024-11-19:
Same kernel panic with that build.

Zormeister, 2024-11-19:
Alright, I'll get to that at some point.

Zormeister, 2024-11-20:
[dmesg-21-11-2024 08.15.54.txt](https://github.com/user-attachments/files/17837160/dmesg-21-11-2024.08.15.54.txt)
[dmesg-21-11-2024 08.47.55.txt](https://github.com/user-attachments/files/17837161/dmesg-21-11-2024.08.47.55.txt)

VisualEhrmanntraut, 2024-11-20:
Don't see any log from the framebuffer before the crash, I would've expected some kind of error. If only we could attach a debugger.

Zormeister, 2024-11-20:
Sucks because the only networking I/O available is the unsupported network card, USB Ethernet would maybe do the trick if I went out of may way to get one that works with Catalina

VisualEhrmanntraut, 2024-11-20:
Doesn't matter because the KDK kernel doesn't boot even though the AMD vanilla patches apply, at least last time I tried. I would guess that there are some additional checks failing.

Zormeister, 2024-11-21:
Random tangent, but the unspecified address led me on a bit of a chase.

The address specified looks to be within the boundaries of the kernel itself, so I decided to disassemble the kernel and try and find the address.

Anyways, from a glance it looks like it's some memset call maybe?

Math:
- KAddr = SlidKAddr - SlideAddr

[image]

VisualEhrmanntraut, 2024-11-21:
I know it's a kernel address, it's at the same range where the IOKit functions in the trace also are. I didn't have the kernel on hand, so I didn't bother attempting to look which function it was. I also theorised that it was some trap handler, rather than something like memset. This is useful information however, thanks.

Zormeister, 2024-11-21:
Maybe, I don't know for sure.

VisualEhrmanntraut, 2024-11-21:
@Zormeister Can you try this?

[NootedRed-1.0.0-RESEARCH_RELEASE.zip](https://github.com/user-attachments/files/17843871/NootedRed-1.0.0-RESEARCH_RELEASE.zip)

Zormeister, 2024-11-21:
Same general panic, however the CR2 value changed to `0x00000000000afc00`

VisualEhrmanntraut, 2024-11-23:
@Zormeister 
[NootedRed-1.0.0-RESEARCH_RELEASE.zip](https://github.com/user-attachments/files/17878725/NootedRed-1.0.0-RESEARCH_RELEASE.zip)

Zormeister, 2024-11-23:
[dmesg-23-11-2024 21.50.45.txt](https://github.com/user-attachments/files/17879389/dmesg-23-11-2024.21.50.45.txt)
[Kernel-2024-11-23-215134.txt](https://github.com/user-attachments/files/17879390/Kernel-2024-11-23-215134.txt)

VisualEhrmanntraut, 2024-11-23:
Well, at least I know the actual function where it’s dying now.

VisualEhrmanntraut, 2024-11-25:
@Zormeister dmesg with this please
[NootedRed-1.0.0-RESEARCH_RELEASE.zip](https://github.com/user-attachments/files/17899899/NootedRed-1.0.0-RESEARCH_RELEASE.zip)

Zormeister, 2024-11-25:
[dmesg-26-11-2024 06.52.49.txt](https://github.com/user-attachments/files/17908259/dmesg-26-11-2024.06.52.49.txt)

VisualEhrmanntraut, 2024-11-26:
```
NootedRed   X6000FB: @ (DBG) AmdDalDmcubService::initializeHardware: cpu_addr = 0x0
NootedRed   X6000FB: @ (DBG) AmdDalDmcubService::initializeHardware: gpu_addr = 0x0
```
Well, here's the problem....

VisualEhrmanntraut, 2024-11-26:
Okay so basically the code is straight up broken here. The DMCUB region is only allocated for Navi 2x. I don't know how it's working in other macOS versions.

VisualEhrmanntraut, 2024-11-26:
Good news is, that I can fix it easily by adding in the missing allocations inside intializeReservedVram (typo's not mine)

VisualEhrmanntraut, 2024-11-26:
@Zormeister It should work now.
[NootedRed-1.0.0-RESEARCH_RELEASE.zip](https://github.com/user-attachments/files/17914317/NootedRed-1.0.0-RESEARCH_RELEASE.zip)

Zormeister, 2024-11-26:
[image]

Confirmed working

VisualEhrmanntraut, 2024-11-27:
Phew, finally.

commit 7d84b90f (2024-11-27): fix: Allocate DMCUB VRAM region for Renoir

Fixes #331
closed 2024-11-27 by VisualEhrmanntraut