ChefKissInc/NootedRed #333: DisplayPort doesn't work on Catalina

Sooly890, 2024-11-29:
### macOS Version

macOS Catalina

### What is the precise OS version on which the bug is occurring?

10.15.7 Release (19H15)

### What is your CPU's model?

AMD Ryzen 5600G

### Describe the unexpected behaviour in detail.

I installed latest MacOS Catalina, plugged in my monitor through DisplayPort, and my computer turned on, but it didn't display anything. I shutdown, plugged into HDMI, and it showed the login screen.

### What should've happened instead?

I install latest MacOS Catalina, plug in my monitor through Display port, my computer turns on, and it displays the login screen.

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

Sooly890, 2024-11-29:
I've just tested Big Sur, it also happens there.

EraserCN, 2024-11-29:
On Monterey here with 3500U, DP doesn't work either.

Sooly890, 2024-11-29:
@EraserCN I'm just installing Monterey now, seeing if it perstsists. I'll get back to this issue.

Sooly890, 2024-11-29:
Yup, indeed, Monterey doesn't work on DisplayPort but is working fine on HDMI.

Sooly890, 2024-11-29:
I just tested Ventura - that works fine.

Sooly890, 2024-11-30:
If it helps - Sonoma and Sequoia are working fine too.

VisualEhrmanntraut, 2024-11-30:
It doesn’t, I don’t have an AMD desktop with integrated graphics, only an AMD laptop with one HDMI port. My main machine is an M3 MacBook Pro.

VisualEhrmanntraut, 2024-11-30:
So, I’m going to spend a some time improving DiskDbg and creating instructions on how to gather the required debugging information so I don’t repeat myself all the time. Although, I must admit that the issue here isn’t really a bug with NootedRed, rather the AMD drivers, but maybe the fix can be back-ported.

VisualEhrmanntraut, 2025-01-13:
https://github.com/ChefKissInc/DiskDbg follow instructions in README. latest log will always be current boot. boot with NRed disabled, install the pkg, enable NRed, boot, then reboot with NRed disabled, fetch the first to last log, and send it here.

Sooly890, 2025-01-13:
Hmm - this is confusing. I installed DiskDbg onto Monterey, Big Sur, and Catalina, however I only managed to get logs from Monterey (I did `launchctl list | grep -v com.apple` and on Catalina and Big Sur and DiskDbg was nowhere to be found). However, NootedRed worked fine on DisplayPort, and the only thing I can think of that changed is my Monitor. I still have it, so I will try again with that one shortly.

[AGDCDiagnose-1736805628.txt](https://github.com/user-attachments/files/18402675/AGDCDiagnose-1736805628.txt)
[dmesg-1736805628.txt](https://github.com/user-attachments/files/18402677/dmesg-1736805628.txt)
[ioreg-1736805628.txt](https://github.com/user-attachments/files/18402676/ioreg-1736805628.txt)

(I have edited the ioreg file, but only to prevent doxing myself)

VisualEhrmanntraut, 2025-01-13:
I need logs only during the black screen. And the ioreg is probably not useful at this moment, so you don't need to attach it for now.

Sooly890, 2025-01-13:
[dmesg-1736807604.txt](https://github.com/user-attachments/files/18402830/dmesg-1736807604.txt)

[AGDCDiagnose-1736807604.txt](https://github.com/user-attachments/files/18402832/AGDCDiagnose-1736807604.txt)

There we go - I'm pretty sure that's the one with the black screen. I knew that the logs before probably were useless, but I posted them anyways in case they were. My boot-args are: `-v debug=0x100 keepsyms=1 revpatch=auto,cpuname,sbvmm alcid=11 -NRedDPDelay -NRedDebug msgbuf=10485760`.

Sooly890, 2025-01-13:
Yes, that's correct. I don't have 2 DisplayPorts on my Motherboard, and it is an AOC monitor.

VisualEhrmanntraut, 2025-01-13:
The dmesg suggests you turned the machine off too early but I'm not sure. Regardless, if it's not working with this monitor, but working with another, that suggests a different problem. What are the specifications of those two monitors? (Resolution, Refresh rate, etc)

VisualEhrmanntraut, 2025-01-13:
I am asking because there might be an artificial bandwidth limitation happening, causing this issue.

Sooly890, 2025-01-13:
I did indeed shut off the computer, but only after it wouldn't output to the monitor after waiting and turning the monitor off and on.

VisualEhrmanntraut, 2025-01-13:
By resolution, I meant in pixels. E.g. My M3 MacBook Pro's pixel resolution would be 3024x1964

VisualEhrmanntraut, 2025-01-13:
Hm, may not be a bandwidth problem then. I'll take a closer look tomorrow as it's 1 in the morning here.

Sooly890, 2025-01-13:
Sure thing

VisualEhrmanntraut, 2025-01-14:
I have pushed a new commit to the DiskDbg code, it should be a bit more stable with the dmesg logging now.

Sooly890, 2025-01-14:
For whatever reason my Xiaomi one didn't work for a few reboots - though I was not able to reproduce it, so I have no logs to share.

Anyhow, I have made some new logs with the new DiskDbg with my AOC monitor:

[AGDCDiagnose-1736892047.txt](https://github.com/user-attachments/files/18416666/AGDCDiagnose-1736892047.txt)

[dmesg-1736892047.txt](https://github.com/user-attachments/files/18416667/dmesg-1736892047.txt)

VisualEhrmanntraut, 2025-01-14:
did you remove msgbuf? cause the dmesg is truncated to 128kb

VisualEhrmanntraut, 2025-01-14:
but from the logs that are here, I do see this:
```
[   11.176938]: [48:0:0] [DAL][EVENT][link:1] >>> type:"Link Training End", status: FAIL, lane_status: 0.
```

Sooly890, 2025-01-14:
Nope, my boot-args remain as `-v debug=0x100 keepsyms=1 revpatch=auto,cpuname,sbvmm alcid=11 -NRedDPDelay -NRedDebug msgbuf=10485760`

VisualEhrmanntraut, 2025-01-14:
Regardless, it seems to be a usual link training failure. Unfortunately the cause is not easy to track down. It could be bandwidth calculations, incorrect timing, race conditions, anything.

Sooly890, 2025-01-14:
Great. I wonder why my Xiaomi one wasn't working for a moment though, will do some more testing at some point.

VisualEhrmanntraut, 2025-01-14:
(and it doesn't help that there's no actual documentation or debugging manual/guide. amd like any other company expects you to use their software as they intend, and just to make a bug report with some pictures and some logs and maybe they look at it and maybe they fix it. I even tried back when NRed had no display signal at all to ask in their mailing list, and they just ignored the message)

VisualEhrmanntraut, 2025-02-24:
> I haven't been able to get DiskDbg working on Catalina or Big Sur however, all the logs I can share are Monterey logs.

It should work now with the latest commit.

VisualEhrmanntraut, 2025-03-22:
Please retry with the latest commit.

VisualEhrmanntraut, 2025-03-22:
Just pushed a fix concerning that panic, thanks for letting me know

PeterParker2008, 2025-05-23:
Probably not the right place/repo/issue thread to write about this. But need help with a critical issue. Possible to DM?
(Will delete this post later)

VisualEhrmanntraut, 2025-11-16:
is this still happening?