ChefKissInc/NootedRed #340: wine craches with gpu errors on Vega 8

soulistyces, 2024-12-19:
### macOS Version

All supported macOS versions

### What is the precise OS version on which the bug is occurring?

15,2

### What is your CPU's model?

AMD Ryzen 5500U

### Describe the unexpected behaviour in detail.

tried several wine applications, Crossover 24.05, Heroic, GamePortingTool 1.1 and beta 3, wineHQ-stable and Devel they all have similar issues wine craches on GPU processing win10 specially steam, opensteamclient, old gog galaxy 1.2 , new gog-galaxy, the only wine game application that i was able to run was ubisoftconnect in win8 crossover 24.05 but with errors, it seeams that divxk vulkan doesnt work well with this ktext nor D3Direct that gives blank and black screen before craching

[image]

[image]

### What should've happened instead?

no GPU or Vulkan divx craches

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-01-13:
Idk about the other apps but I'm pretty sure Steam's client is a web-based application, so it's probably the same blend function problems as #158

soulistyces, 2025-01-13:
i figured out its amd hackintosh related, tested with a laptop intel with an gtx 1070 and doesnt crach, tested with a ryzen 3600 with an rx580 8gb and also crachs but only downloading games when it finished the forced reopen steam several times to continue download it doesnt crash and plays the game....dont know why but its amd hacking tosh related

closed 2025-02-17 by VisualEhrmanntraut