ChefKissInc/NootedRed #344: few latest Nootdred versions NOT use HW accel for iGPU

dipol0, 2025-01-12:
### macOS Version

macOS Sonoma

### What is the precise OS version on which the bug is occurring?

14.7.1

### What is your CPU's model?

AMD Athlon 3000G

### Describe the unexpected behaviour in detail.

few latest Nootdred versions NOT use HW accel for iGPU

### What should've happened instead?

Mobo ASUS ASUS PRIME A320M-K (BIOS version6232)

Use Opencore v.1.0.3 and few latest Nootdred versions NOT use HW accel for iGPU (AMD Radeon Vega 3). Boot normal, but not HW accel and sysinfo show only 7Mb VRAM. I try oldest versions Nootedred (downloaded 7 apr 2024) and HW accel work and show all 2Gb VRAM (i use resizebar enabled in BIOS). Latest versions not supported old CPU now? Or what?

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-01-12:
Update Lilu...

closed 2025-01-12 by VisualEhrmanntraut

dipol0, 2025-01-12:
> Update Lilu...

Tnx! Worked. But some times get reboots (on boot process) or black scerren. Looks like old bug or my config have not all compatability with hakintosh.