ChefKissInc/NootedRed #347: The time to enter the desktop at startup becomes very long

diandiweb, 2025-03-01:
### macOS Version

macOS Ventura

### What is the precise OS version on which the bug is occurring?

13.6 

### What is your CPU's model?

AMD Ryzen 9 5900HX

### Describe the unexpected behaviour in detail.

The time to enter the desktop at startup becomes very long, taking about 30 seconds to one minute

### What should've happened instead?

Approximately 10 seconds display time

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-03-02:
You have not answered the question about logs and for this particular kind of issue it is next to impossible to understand what the cause would be without them. Please attach related logs for this issue as was asked by the issue template. Failure to do so will lead to closure of this issue.

VisualEhrmanntraut, 2025-03-22:
No response, cannot proceed further. Please also test the latest commit.

closed 2025-03-22 by VisualEhrmanntraut