ChefKissInc/NootedRed #348: Kernel Panic from NootedRed

Hendrik292, 2025-03-08:
### macOS Version

macOS Ventura

### What is the precise OS version on which the bug is occurring?

14.7

### What is your CPU's model?

AMD Ryzen 5 3500U

### Describe the unexpected behaviour in detail.

I try to boot into the installer and it got stuck

### What should've happened instead?

The installer should open

### Attach any logs related to this issue. For example .gpuRestart, .panic, etc files.

[image]
[image]

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-03-08:
Toggle Legacy Boot/CSM

Hendrik292, 2025-03-08:
Thank you it worked

closed 2025-03-08 by Hendrik292