ChefKissInc/NootedRed #354: NootedRed causes a kernel panic in macOS 15.4

azu0609, 2025-04-02:
### macOS Version

macOS Sequoia

### What is the precise OS version on which the bug is occurring?

15.4 (24E248)

### What is your CPU's model?

AMD Ryzen 5 5500U

### Describe the unexpected behaviour in detail.

After updating to the 15.4 from 15.3.1, the system kernel panics at boot with NootedRed Enabled. It boots normally without NRed.

### What should've happened instead?

The system boots normally without a kernel panic.

### Attach any logs related to this issue. For example .gpuRestart, .panic, etc files.

[Kernel-2025-04-02-103822.zip](https://github.com/user-attachments/files/19577190/Kernel-2025-04-02-103822.zip)

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-04-03:
This panic log is very confusing, it is not immediately clear what exactly wrong, the location is somehow seemingly random-looking. I will try to investigate.

VisualEhrmanntraut, 2025-04-06:
Something weird in Lilu's function routing was going wrong I guess and it was fucking things up

commit d29f7ce8 (2025-04-06): fix: Bug fixes and cleanup related to Patcher+

Fixes #354

Had to revert one of the improvements because Lilu is missing some null checks among some weird behaviour in it that was causing 15.4 not to work with the firmware injection code.
closed 2025-04-06 by VisualEhrmanntraut