ChefKissInc/NootedRed #361: RDNA 2 iGPU (Zen 4) Support for Hackintos

veys91, 2025-04-28:
### OS

All supported macOS versions

### What is the precise OS version on which the feature is missing?

14.5

### What is your CPU's model?

Ryzen 7 7700

### Describe the feature in detail.

Hello,
I am using a Ryzen 7 7700 (Zen 4) processor with an integrated RDNA 2 GPU (iGPU) and would like to request official support for RDNA 2 iGPUs in Hackintosh setups. Currently, there is limited or no native support for RDNA 2 iGPUs in macOS, which makes it challenging to fully utilize the hardware acceleration and graphics capabilities that these integrated GPUs offer.

Despite using the AMDRadeonX4000.kext and other necessary patches, I am experiencing issues with display output, stability, and performance. I believe that with proper support for RDNA 2 iGPUs, users with Zen 4 APUs like mine could have a much smoother experience in Hackintosh systems.

Therefore, I am formally requesting Nooted Red and the Hackintosh community to consider adding support for RDNA 2 integrated graphics, so that users can fully take advantage of these newer processors and integrated GPUs. This would not only improve the Hackintosh experience but also provide better compatibility for those of us using AMD's latest Ryzen APUs.

Thank you for your attention and consideration.

### Have you tried any alternatives? If so, what are they?

no...

### Issue Submission Terms

- [x] You have made sure that there are no other requests about this feature.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-04-28:
> would like to request official support

for free? no

closed 2025-04-28 by VisualEhrmanntraut

ihiuhi65985, 2025-06-14:
Could you please share your efi? I have been setting it up for a long time but cannot enter the download screen

RoboCrafty, 2025-06-27:
> > would like to request official support
> 
> for free? no

How much money do you want for it? @VisualEhrmanntraut xD

Im sure people like me and the whole zen 4 community can easily crowdfund it