ChefKissInc/NootedRed #362: support RDNA3 780M？

lky005, 2025-05-09:
### OS

All supported macOS versions

### What is the precise OS version on which the feature is missing?

15.2

### What is your CPU's model?

AMD 7840HS

### Describe the feature in detail.

support RDNA3 780M？

### Have you tried any alternatives? If so, what are they?

no

### Issue Submission Terms

- [x] You have made sure that there are no other requests about this feature.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

closed 2025-05-09 by VisualEhrmanntraut

RoboCrafty, 2025-06-12:
@VisualEhrmanntraut When do you think realistically v2.0 would come out? Brining support for 8845hs and the likes?

lky005, 2025-06-14:
i dont know

WhySooooFurious, 2025-08-05:
This would be great for my 8845hs!

Let's hope support comes soon.