ChefKissInc/NootedRed #366: Kernel Panic On Tahoe

taqieallah, 2025-06-11:
### macOS Version

All supported macOS versions

### What is the precise OS version on which the bug is occurring?

26.0 Beta

### What is your CPU's model?

Ryzen 7 5850u

### Describe the unexpected behaviour in detail.

Im Getting kernal panic while loading the kernel in tahoe 

### What should've happened instead?

should boot normaly

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-06-12:
I've already gotten it working immediately after OpenCore got fixed. Acidanthera is just being slow as usual and not merging the constants PR for Lilu.

[NootedRed.kext.zip](https://github.com/user-attachments/files/20705062/NootedRed.kext.zip)

elfametesar, 2025-06-13:
> I've already gotten it working immediately after OpenCore got fixed. Acidanthera is just being slow as usual and not merging the constants PR for Lilu.
> 
> [NootedRed.kext.zip](https://github.com/user-attachments/files/20705062/NootedRed.kext.zip)

well ig the good thing is that the issue with some shader functions causing freezes, corruption and freezes seem to be almost gone. the regular stuff that caused issues on sequoia is fine on tahoe. i only got that issue once on photoshop ai generation subscription page.

vebhar, 2026-05-29:
I am facing something strange even stable version of nooted red works on sequoia , but fails on tahoe, does nooted rid zip above here is different than current release vesion which is i guess at 0.8.0 / beta one 0.9.0, which one will be better , i am doing some kernel research work and using x512da asus vivobook 15 , amd 3500 u , vega 8 , 8gb ram, 2gb graphics integrated.