ChefKissInc/NootedRed #380: Kernel Panic when booting a first-time install of Monterey

keneshindev, 2025-07-19:
### macOS Version

macOS Monterey

### What is the precise OS version on which the bug is occurring?

12.7.4 (21H1123)

### What is your CPU's model?

Ryzen 3 5300U

### Describe the unexpected behaviour in detail.

Booted to the macOS installation after macOS Installer reboots, resulted in a KP

### What should've happened instead?

The system should've booted normally to OOBE

### Attach any logs related to this issue. For example .gpuRestart, .panic, etc files.

[kernel.txt](https://github.com/user-attachments/files/21328019/kernel.txt)
From PanicInfoReader

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-07-19:
That's weird. Thanks. I'll try to take a look when I get the time.

keneshindev, 2025-07-20:
[AMDRadeonX5000.kext.zip](https://github.com/user-attachments/files/21334298/AMDRadeonX5000.kext.zip)
Will this do?

VisualEhrmanntraut, 2025-07-20:
No, I need *all* the ones that start with AMDRadeonX5000

keneshindev, 2025-07-20:
GitHub won't let me upload it here, as it's too big
Although maybe i shouldn't have included the .bundle files
https://drive.google.com/file/d/1DVN-4MitOVSMAEv55H3dNkppvZutHzkn/view?usp=sharing

VisualEhrmanntraut, 2025-07-20:
does it work with the commit I just pushed?

keneshindev, 2025-07-20:
Yep, it does boot without a KP and has hardware acceleration
Firefox seems to not work, but i'm not sure if that's related as it didn't work with VESA either

VisualEhrmanntraut, 2025-07-20:
It's not related

closed 2025-07-20 by VisualEhrmanntraut