ChefKissInc/NootedRed #384: Graphical artifacts occurred in Safari when using GPU in a high intensity way

Mike-Luo-Leap, 2025-08-08:
### macOS Version

All supported macOS versions

### What is the precise OS version on which the bug is occurring?

14.7.5

### What is your CPU's model?

3.2GHz AMD Ryzen 7 5800H

### Describe the unexpected behaviour in detail.

I opened Bilibili, then watched a video, then graphical artifacts occurred.
For some reason of privacy, I blocked some information of the website.

[image]

[image]

### What should've happened instead?

I don't think this bug needs to fill this blank.

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

Rutrama, 2025-08-09:
This seems a lot like the same blend function issue we already experience with Chromium and (rarely) Firefox. I tested bilibili on Safari 16.6 (MacOS Ventura 13.7.7) and Firefox for good measure. It played a video without issues, so maybe it's a Sonoma or recent Safari update issue? This could be concerning, since Safari is the fallback for those few websites that are unusable in Firefox.

hrx114514x, 2025-08-09:
> This seems a lot like the same blend function issue we already experience with Chromium and (rarely) Firefox. I tested bilibili on Safari 16.6 (MacOS Ventura 13.7.7) and Firefox for good measure. It [...]

How you use safari to watch videos？Safari cannot disable gpu accelerating,and some video in bilibili become green and I have to use edge.

MinecraftNerrd1337, 2025-08-27:
Im sorry but using Edge on a Mac... I can't let that slide without saying anything 🤣
Shouldn't Brave/Chromium work when you disable hardware acceleration?

hrx114514x, 2025-08-30:
> Im sorry but using Edge on a Mac... I can't let that slide without saying anything 🤣 Shouldn't Brave/Chromium work when you disable hardware acceleration?

Yes，but I wonder how he play video without issues in Safari,because no hardware acceleration settings in safari.Edge can run.