ChefKissInc/NootedRed #386: Limited Metal Support on Vega 7

BSD-gu, 2025-09-04:
### macOS Version

macOS Sequoia

### What is the precise OS version on which the bug is occurring?

15.6.1

### What is your CPU's model?

AMD Ryzen 5625U with Vega 7

### Describe the unexpected behaviour in detail.

I installed the NooTedRed Kext on an AMD Hackintosh with Ryzen 5625U + Vega 7 to enable GPU acceleration. While the desktop environment is smooth and 4K videos play without issues, iOS simulators remain laggy and unusable:

Scrolling and opening apps in simulators is
Super laggy.

Older simulator devices (iPhone SE, or iphone 8) perform slightly better but still lag.

This lag occurs across multiple Xcode versions (tested 16.2 and older).

Even though Metal 3 is reported as supported, iOS simulators appear to partially fall back to CPU/software rendering, which causes the performance issues.

### What should've happened instead?

iOS simulators should run smoothly, fully leveraging Metal 3 and GPU acceleration.

### Issue Submission Terms

- [x] You have made sure that there are no other reports about this bug.
- [x] You do not use configurator software (e.g. OpenCore Auxiliary Tools, OpenCore Configurator).
- [x] You have made your own EFI according to the ChefKiss Hackintosh Guide.
- [x] You are not using any software that may conflict, including, but not limited to, OpenCore Legacy Patcher, BFixup, WhateverGreen.
- [x] You acknowledge that violation of these terms may result in your issue being closed or your user being blocked from the organisation.

VisualEhrmanntraut, 2025-09-05:
if there are blur effects in the simulator it is running entirely on GPU and any slowdown is because of cpu limitations (e.g. clock running too slow), a bug in apple's software, or the known issues

BSD-gu, 2025-09-05:
I was monitoring the processor frequencies, which spiked up to 4.2 GHz while running the simulator (the maximum frequency for this chip). So, it’s definitely not a CPU power-saving issue. I did notice some blur, like the bottom dock in the simulator, but the sluggish performance makes me suspect it fell back to software rendering. I believe this CPU is more than capable of handling iOS simulators easily, so the problem likely lies elsewhere.

VisualEhrmanntraut, 2025-09-05:
software rendering does not support blur effects on things like the dock so it's definitely not that.

BSD-gu, 2025-09-05:
I might be wrong, but I ran some tests on the iOS 16 Pro Max simulator. Initially, it was extremely laggy and almost unusable. Then I changed the home screen wallpaper to solid black, which somehow made it slightly less laggy. Could it be that hardware acceleration with NootedRed is limited?

BSD-gu, 2025-09-05:
Swiping left and right through the apps spikes CPU usage to around 40–50%, while idle usage (with iOS simulator running) sits at about 20%.